Ultimaker/Cura #9227: Sheen not uniform if slicer doesn't print consecutive travels - requesting finish / sheen settings

JerichoGuitars, 2021-02-02:
[https://imgur.com/cYihrQY](url)
The issue is when the travel skips a section and comes back to it, the light reflectivity changes on the part if time is given for it to extrude and then come back to fill in the parts missed.  It goes like an Alleyway game, skipping the important parts.

If the nozzle leaves at least 1-2mm off the print and then comes back to fill, regardless of the geometry of the part, the top sheen will be uniform and require less post-processing.

This affects all users who are using prints commercially.

fvrmr, 2021-02-04:
Thank you @JerichoGuitars for your feature request. 
We already got this request a few times. So I will close this issue because it is a duplicate of #8909

closed 2021-02-04 by fvrmr

JerichoGuitars, 2021-02-04:
I don't see that it has been added to Cura, I don't understand how it could
not be prioritized, its costing us so much money.  Can we pay for it to be
prioritized?

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

fvrmr, 2021-02-05:
Unfortunately you can't donate money to prioritize feature requests for us. Hope you understand.

Ghostkeeper, 2021-02-09:
We do accept pull requests though, even if those pull requests are made by a developer who was paid for it. (How would we even know?)

JerichoGuitars, 2021-02-09:
Will you guys be adding the Monotonous infill feature like Prusa has?

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

fvrmr, 2021-02-09:
As you can see in the other issue, we've created a ticket for this so it is on our backlog. But it is not prioritized. So I can't give you a timeline when we will implement this.