Ultimaker/Cura #9228: [FR] someway to insert filament custom G-code or filament information

ChipCE, 2021-02-02:
**Is your feature request related to a problem? Please describe.**
I have many machine with difference specs, so I must manage filament profile for each machine. <code>ex: MachineA_PLA, MachineA_ASB, MachineB_PLA, MachineB_ABS</code>. The more type of machine(or extruder) user had, the more filament(or printing profile) they have to manage.

**Describe the solution you'd like**
If there is a way to insert custom g-code based on filament <code>Ex: FILAMENT_TYPE = PLA</code>, each printer can handle other parameter setting using macro(RRF and klipper). There is <code>{material_guid}</code> on the replacement patterns but that value is auto generated and hard to manage in multi cura installation.

**Describe alternatives you've considered**
Someway to insert filament-based g-code or a filament replacement patterns that user can set in the material setting. 

**Affected users and/or printers**
Any firmware that support macros

fvrmr, 2021-02-04:
Hi @ChipCE thank you for your feature request. 
I will have to defer this issue, because you can achieve this by material profiles in Cura. Also we don't maintain these firmwares. 
But a pull request is always welcome, so other users that uses those firmwares can use it too. 

That why I will close this issue, hope you understand!

Ghostkeeper, 2021-02-09:
The material profiles can override the extruder start and extruder end g-codes, but not the global start and end g-codes (since you can have different material profiles in each extruder). There is no interface element for it though, so it involves editing the .xml files.

closed 2023-07-13 by github-actions[bot]