Ultimaker/Cura #9230: macOS big SUR Version 11.0.1 cannot rotate or move imported sty on build plate.

noahangel11, 2021-02-03:
**Application version**
Cura 4.8.0

**Platform**
macOS big SUR Version 11.0.1 

**Printer**
Ender 3 Pro

**Reproduction steps**
1. Import an STL
2. I cannot select the sty image on the build plate so that I can rotate or remove. This seems intermittent. Sometimes I can, sometimes I cannot, with the same file. I have tried multiple files

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
I cannot rotate or move object. Selecting it from object list will bring up the rotate manipulators but i cannot click on them to move object. Mouse selection causes manipulators to disappear.

**Expected results**
I should be able to manipulate the position.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

jellespijker, 2021-02-03:
Duplicate of #8745.
I will close this issue.

closed 2021-02-03 by jellespijker

noahangel11, 2021-02-03:
So there is no solution yet?

jellespijker, 2021-02-03:
Not yet but very high on our priority list.