Ultimaker/Cura #9234: Filament Runout Sensor do not work via USB - Ender 5 Plus - Cura 4.8 Big Sur

aquajazz, 2021-02-03:
Hi folks,

The Filament Runout Sensor is working with SD-Card print from Cura, but not with USB-Print via Cura.

**Application version**
4.8.0
Platform MacOS 11.1 Big Sur
MacBook Pro 2019
2,6 GHz 6-Core Intel i7
16GB of memory

**Printer**
Creality Ender 5 Plus
Ver. 1.70.1 BL
Silent Board
Creality Metal Extruder
PTFE

Reproduction steps
print stl-file via SD-Card
cut filament just befor sensor
print stl-file via USB
cut filament just before sensor

**Actual results**
SD-Card: when filament runout sensor light switch off, print-job pauses until I react on the request of filament change on the display. I`m able to resume printing with a new roll of filament.

USB: When the sensor detects, that filment runs out, sensor light switches off. But the print-job do not stop, it is continued without filament.

**Expected results for USB print-jobs**
If printer runs out of filament:
    message via Cura or printer display
    printer switch to "pause"
    resume print job after the filament change

**Project file**
Any stl-file sliced by this Cura version.

Thanks,
#Thomas

fvrmr, 2021-02-04:
Hi @aquajazz thank you for your report.
This is not Cura related, but it has to do with your firmware. And since you are using a Creality, I have to defer this issue because we don't maintain them.
Feel free to make a pull request to solve this problem. 
That's why I will also close this issue.

closed 2021-02-04 by fvrmr

Ghostkeeper, 2021-02-10:
It's not something that Cura can really solve as far as I now. See my explanation in the previous bug report about this: https://github.com/Ultimaker/Cura/issues/6036#issuecomment-776659881