Ultimaker/Cura #9235: UI to large to view on load

uglygargoyle, 2021-02-04:
Application version - 4.8.0
Platform - Windows 10
Printer - Creality Ender 3
Reproduction steps**
I opened Cura on my PC yesterday (v4.6) and got the below screen instead of the normal UI it almost looks like it has zoomed right in. I installed v4.8 and it worked fine for the day but I have started it again today and got the same screen. I have checked my graphics card drivers as all good. Im using 2 monitors so disabled one and tried but the same result.

Screenshot(s)
[image]

[cura.log](https://github.com/Ultimaker/Cura/files/5926509/cura.log)
Log file attached

fvrmr, 2021-02-04:
Hi @uglygargoyle 
In your log file stands this: `2021-02-04 13:41:10,633 - ERROR - [MainThread] UM.View.GL.OpenGL.__init__ [58]: Startup failed due to OpenGL context creation failing` so it means that your computer can't access OpenGL

Ghostkeeper, 2021-02-12:
Since it apparently worked on the same computer previously, something must have changed on the software and broke Cura. I can think of two things which could have a result like this:

* You've updated your graphics drivers and the new update has a regression. You could try rolling back your graphics drivers then, though that is not something most people would like to do. Another option would be to go into Cura's preferences and enable the compatibility mode for layer view, which only initialises OpenGL 2 rather than 4. If going into the preferences is impossible in the interface due to this bug, there is also a configuration file in `%APPDATA%/cura/4.8/cura.cfg` that can be edited with a text editor. You'd set `force_layer_view_compatibility_mode = True` under the `[view]` category.
* Some shader cache got corrupted, perhaps due to force-quitting Cura, power loss, etc. There is also a `qtshadercache` folder under `%APPDATA%`, iirc. It can be deleted safely to reset the shader cache of all your Qt-based applications.

uglygargoyle, 2021-02-12:
I uninstalled cura and deleted all of the folders and then rebooted and reinstall and it worked and has worked since so I have closed this issue. Thanks to all for comments

closed 2021-02-12 by uglygargoyle