Ultimaker/Cura #9237: Feature Request: Overhanging Wall Speed Affects Layer/Layer Range

juspch, 2021-02-04:
Hello everyone, i read about my issue on a previous post but OP had considered his issue resolved with the addition of the experimental overhang wall speed setting. I am here to sort of revive the issue again as I believe the current implementation of the setting is lacking. 

 

I am starting to find that the Overhang Speed option does not work well when dealing with actual overhang curling at corners no matter how low I go with the setting. My leading hypothesis on why this happens is that my overhanging corners don't have the sufficient length such as to produce enough additional layer cooling time, while slowing down my print speed too much at the overhang instead has the alternative effect of adding too much heat and causing excessive sagging or sticking to the nozzle. Because of this, I am forced to slow down my entire print instead of just those overhang layers or tediously find the overhangs using the Overhang Speed setting using Preview view and manually reducing speeds for those layers using ChangeatZ.

 Already posted this on the forums, reposting here as it felt like a more appropriate venue.

In light of this, I think it would be an improvement to the current software if overhanging wall speed could be set to affect the entire layer or a range of layers for which the overhang of a certain angle was detected. Likewise, if it is found that overhang curling is more directly related to layer cooling time, then perhaps a setting for minimum layer time could be implemented instead of a speed percentage.

My other opinion of as to why curling happens on overhanging corners is that residual tension from the PLA pulls up overhangs on corners as these have no layers underneath them to grip on to. This residual tension is worsened as print speed is turned up, extrusion becomes less accurate, and the PLA being printed out becomes more stretched out onto the last layer. I have not yet tested this as I do not really know how to, but if this is the case, another better implementation of the overhanging wall speed setting would be to have a setting for an additional length for which to apply the overhanging wall speed too. For example, instead of just having 50% print speed for the overhang, the speed reduction would extend for 1 mm before the start of the overhang and 1 mm after the end.

TLDR: I feel Overhang Wall Speed can be improved by adding options to apply the speed reduction to more than just the overhang area itself such as for a certain distance before and after the overhang or even to the entire layer/ a range of layers containing the overhang.

fvrmr, 2021-02-05:
Hi @juspch thank you for your feature request. I will discuss it with the team. Keep you posted!

fvrmr, 2021-02-08:
I have discussed it the team. We will defer this feature request, it will take a lot of time implementing this because we think not many users will use it. 
Hope you understand!

darthdively, 2021-02-14:
> I have discussed it the team. We will defer this feature request, it will take a lot of time implementing this because we think not many users will use it.
> Hope you understand!

Please do not differ this. Many of us are suffering from this. I understand that there will be a lot of time invested in such a feature. Thank you very much for all the hard work the team does.

Ghostkeeper, 2021-02-24:
We keep track of these issues, to be sure. But we have to defer like 75% of them to be more realistic about what we can spend time on. Often they get implemented aside with something else years down the line. Sometimes the priority changes and we'll build it anyway. Sometimes the need for it just disappears entirely, e.g. if we were to integrate the bridging routine with normal overhang.

In this case the decision was weighed towards deferring, because:

* Flow changes are problematic here, but there are other features in the works that might also solve that issue for you.
* If we add a setting like this, it will be pretty hard to understand the setting. You'd get a setting name like "Overhang Speed Change Advancement" which would explain that it makes the speed change for overhang extend a bit beyond the region of the overhang. This is pretty complex to visualise for most people, especially if they didn't read this thread. As a result, very few people will be using the feature.
* If people use the feature incorrectly, it can lead to print quality problems (ringing) and confusion as to why the nozzle is slowing down in a seemingly random location completely disconnected from overhangs.
* Making a fairly direct implementation (break up a line in a certain spot) would not be very maintainable. To make a maintainable implementation would be a lot of work.

So basically we realise it's an issue and your suggested feature might be a solution, but also has problems and considering the amount of work we don't think it brings enough of an improvement to enough people.

closed 2023-07-13 by github-actions[bot]