Ultimaker/Cura #9238: Problem with Ultimaker Cura, print start immediately and don't wait pre-heat Extruder and build plate

Bighoneypot, 2021-02-04:
I try to use a Ultimaker Cura with my Ender 3 Pro with marlin and Makerbase Robin nano V1.x with WiFi.

All work fine, but I have a problem time to preheat

The problem is when I send my gcode over wifi to my printer and print start immediately and don't wait pre-heat Extruder and build plate.

This is my Start Gcode

; Ender 3 Custom Start G-code new
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

and this my EndG-code
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positionning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

juspch, 2021-02-05:
If I were you, I'd open up the gcode itself and see what M104 commands specifically are being sent to the printer.

nallath, 2021-02-08:
Is this only when you send it over wifi? How does the g-code actually look?

GregValiant, 2021-02-09:
My Start Gcode is nearly identical to yours.  The snippet is from an actual gcode file.  The heating lines are inserted by Cura prior to the Start_gcode.  (I removed the Start Gcode line "M190" as I don't need it for PLA)

;Generated with Cura_SteamEngine 4.8.0
M140 S50
M104 S210
M105
M109 S210
M82 ;absolute extrusion mode
**; Ender 3 Custom Start G-code**
G92 E0 ; Reset Extruder
G28 ; Home all axes
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

Bighoneypot, 2021-02-09:
Thank you, I'm try this..

Bighoneypot, 2021-02-16:
Don't work.

GregValiant, 2021-02-16:
Copy and paste these 4 lines into your start Gcode.  They go right after the G28 line.

M140 S{material_bed_temperature_layer_0} ;Start heating the bed
M104 S{material_print_temperature_layer_0} ;Start heating the hot end
M109 S{material_print_temperature_layer_0} ;wait for the hot end
M190 S{material_bed_temperature_layer_0} ;wait for the bed

But post one of the problem Gcode files here as well.  You really shouldn't need the above lines.  Cura should do it.

Ghostkeeper, 2021-02-17:
I'm also still wondering how the resulting g-code file looks that Cura generates. Could you show a snippet and/or project file? It's in the issue template, and we kind of need it to be able to reproduce your problem.

Bighoneypot, 2021-03-03:
I have to do another test, tomorrow I send the answer

Bighoneypot, 2021-03-15:
I tried, sometimes it works sometimes it doesn't. If I send large print files it does not transfer the file correctly and does not print

Ghostkeeper, 2021-03-29:
I'll ask again. Could you show a g-code snippet from the start of the generated file, and/or a project file?

As it stands, I cannot reproduce this issue, as you'll understand. My printer pre-heats the extruder and build plate fine. I can't fix anything if I don't know how the issue is reproduced.

closed 2021-04-12 by no-response[bot]