Ultimaker/Cura #9240: More Ironing Presets besides "zig-zag and concentric"

JerichoGuitars, 2021-02-05:
**Is your feature request related to a problem? Please describe.**
zig-zag creates inconsistent surface over all parts produced for sale. partly from the way that it irons in different places throughout the whole top surface, not like an iron at all, actually.  Concentric leaves visible marks in the center of the pieces, if the pieces have holes in them it becomes even worse.

More options for Ironing

**Describe alternatives you've considered**
Selling my Ultimaker S5 seems to be the only way to remedy the situation, but my company is already very invested in this machine.

**Affected users and/or printers**
Anyone who prints various items for guitars at a commercial level.

fvrmr, 2021-02-05:
Hi @JerichoGuitars what kind of presents/options you have in mind?

JerichoGuitars, 2021-02-05:
I have created a video here;
https://imgur.com/A8Hy9Gd
I will attach a few others to another email.

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

JerichoGuitars, 2021-02-05:
https://imgur.com/j5rJt6M

This one is much like Monotonous Infill that many happy users enjoy on the
prusa slicer
https://youtu.be/s9rrLDRyMxU

There are many examples of how it makes a usable part without post
processing.

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

Ghostkeeper, 2021-02-12:
Monotonic ordering is a duplicate of this issue: https://github.com/Ultimaker/Cura/issues/8909
I'd still consider it a lines pattern, but an option part of that.

The other ordering seems to go back and forth a bit, skipping a line and then going back to the skipped line. What do you reckon would be the point of doing this?

JerichoGuitars, 2021-02-13:
Mono tonic infill isn’t a feature it is a capability. I have asked some people familiar with polymers/monomers/thermoplastics, my dad was always talking about it when he was VP of chem side at Sherwin Williams but I never fully understood until I recently bought 2 Ultimakers for my company. Amazing machines, we made the right choice. 

If you have varying degrees of temp surrounding materials closer to glass transition then regardless of dimension reflectivity is not uniform. If the extruded filament is 200c coming out, the filament next to it is effecting it’s reflectivity. If every strand is laid where the left side it is 120c, the bottom is 85c, and the right is not extruded, this produces different reflectivity than if it is 90c on left side, bottom is 85c, and the right is 120c. 

By varying temperatures the same material can be more/less glossy which effects reflectivity. This is why extrusion temp effects gloss and uniformity is typically on Z axis wall.  This is a case for mono tonic extrusion as a capability. Without it, any machine is inferior to the one with the greater capability. 

Therefore;

If you extrude a filament where uniformity exists in any form In the 4th dimension, you create a greater resulting uniformity. Extruding at 200c for instance, If the bottom of the filament laid is 100c and all other sides are open air, then only one side becomes 180c, the longer you maintain this property over time (the 4th dimension) you attain uniform reflectivity.

JerichoGuitars, 2021-02-13:
Mono tonic infill isn’t a feature it is a capability. I have asked some people familiar with polymers/monomers/thermoplastics, my dad was always talking about it when he was VP of chem side at Sherwin Williams but I never fully understood until I recently bought 2 Ultimakers for my company. Amazing machines, we made the right choice. 

If you have varying degrees of temp surrounding materials closer to glass transition then regardless of dimension reflectivity is not uniform. If the extruded filament is 200c coming out, the filament next to it is effecting it’s reflectivity. If every strand is laid where the left side it is 120c, the bottom is 85c, and the right is not extruded, this produces different reflectivity than if it is 90c on left side, bottom is 85c, and the right is 120c. 

By varying temperatures the same material can be more/less glossy which effects reflectivity. This is why extrusion temp effects gloss and uniformity is typically on Z axis wall.  This is a case for mono tonic extrusion as a capability. Without it, any machine is inferior to the one with the greater capability. 

Therefore;

If you extrude a filament where uniformity exists in any form In the 4th dimension, you create a greater resulting uniformity. Extruding at 200c for instance, If the bottom of the filament laid is 100c and all other sides are open air, then only one side becomes 180c, the longer you maintain this property over time (the 4th dimension) you attain uniform reflectivity.  

https://polymerdatabase.com/polymer%20physics/Gloss.html

smartavionics, 2021-02-14:
Hi @JerichoGuitars , I have put together a simplistic implementation of the monolithic skin pattern, here's a bad image of an example print, top is original lines pattern, bottom is monolithic lines pattern...

[image]

If you want to try it out and are willing to install one of my builds, I can make a release for you. What system are you using? I can build for all platforms except for MacOS running Big Sur.

JerichoGuitars, 2021-02-14:
Man it looks better to me, what do you think?  Physics works!  If you were to add many pattern options or the ability to pick a pattern based on a selection, etc, I have many ideas for patterns. 

Can you possibly try another print with no walls or all infill first, single wall at end of layer vice versa?

I am running Cura on Catalina

smartavionics, 2021-02-14:
Here's the same print with no walls...

[image]

I'll make a MacOS build that you can try.

JerichoGuitars, 2021-02-14:
In terms of uniformity it’s fantastic. I do think extrusion pattern “lines” and “zig zag” should operate this way.  it would be great if I could make changes to it and send them to you. Would also be great as ironing function at end of print and between each layer as well. Thank you. With flow changing and some tweaking cura would have the best sheen and quality prints by a long shot

smartavionics, 2021-02-14:
At the moment, the monotonic printing only works with the lines pattern and not zigzag or concentric. I can't imagine right now how it could work with those patterns. Anyway, I'm just building a MacOS release for you to try, it will be cooked soon...

smartavionics, 2021-02-14:
See https://github.com/smartavionics/Cura/releases/tag/20210214

This has a Monotonic Skin Lines setting that is visible when the skin pattern is "lines".

My builds can be installed alongside the UM builds, they shouldn't conflict at all.

JerichoGuitars, 2021-02-14:
Amazing I will get this first thing! Thank you. 

I would imagine you could ask for a coordinate for concentric which would generate a point on or off the plate which could generate the path for infill and the extruder would follow that path, extruding only over the model. Tweaking in preview mode you could find one of a kind surface finishes each time. A Z axis spline or line (2 points) in 3D space would generate the tool paths for each slice. 

Example for a one foot cube in middle of build plate would be to choose [45,146,5] and [88,33,111] to generate a concentric slicing pattern which would infill without regard to holes in the model, like your print. Arrange your print on the plate so the center is not over top of the print anywhere and you would then be adding a sort of “angular luster” to your print. 🇺🇸🤘

JerichoGuitars, 2021-02-14:
[image: Screen Shot 2021-02-14 at 10.34.31 AM.png]

Catalina has been creating havoc, not sure if this is related or not.

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

smartavionics, 2021-02-14:
Your image doesn't show because github ignores images in emails, you need to attach it here.

JerichoGuitars, 2021-02-14:
Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

smartavionics, 2021-02-14:
Sorry, I should have said earlier that if you have any problems with my Cura build, please open an issue at https://github.com/smartavionics/Cura rather than discussing here, thanks.

JerichoGuitars, 2021-02-14:
Mark,

Not sure what you mean?  It says it is damaged and cannot be opened.  I
would love to try it out.

Thank you,
Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

smartavionics, 2021-02-14:
What I mean is don't discuss problems with my builds here, open an issue at my repo, the UM devs don't want to be spammed with stuff that isn't relevant to them.

JerichoGuitars, 2021-02-14:
Ok you mean once i can open it if I have issues with it I would have to
uninstall it or contact your repo for any issues with it?

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

smartavionics, 2021-02-14:
How many fucking times do I have to say it, if you have ANY problems with my releases, open an issue on my repo. That's it, I'm tuning out of this issue now, goodbye...

JerichoGuitars, 2021-02-14:
I haven't yet been able to access any of them??

Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705

JerichoGuitars, 2021-02-14:
I don't know how you made your way into my inbox but this is a thread for
an issue with Ultimaker Cura, that's who I thought I was talking to on
account of the email thread attached.  I'm a company owner and I have
opened an issue with the Cura software for our commercial products.

Have a good day,
Dave Muselman
Jericho Guitars, LLC
jerichoguitars.com
baritoneguitars.com
(866) 648-4827 x 705