Ultimaker/Cura #9242: Cura 4.8 ads additional support

3dparade, 2021-02-05:
Hello,

we have a support issue on Cura 4.8.
The model is a tube-shape, with vertical wall, which should not be supported at all.
But Cura 4.8 is adding support columns around it. We have tried the support blocker solution,
but without any good result. In spite of the support blocker, the columns are still there.
[image]
[image]

Thanks for help in advance.
Gábor

fvrmr, 2021-02-05:
Hi @3dparade could you share your project file? So I can have a look?

3dparade, 2021-02-05:
> Hi @3dparade could you share your project file? So I can have a look.
> Did you try to change the `support overhang angle` already?

Hi fvrmr,

thanks for reply. Yes, I have tried it... the support disappears if the value is 90, but in that case all support are gone, even
the needed one.

fvrmr, 2021-02-05:
Can I take a look at your project file then?

3dparade, 2021-02-05:
[UM3_cura4.8_issue.3mf.zip](https://github.com/Ultimaker/Cura/files/5933511/UM3_cura4.8_issue.3mf.zip)

fvrmr, 2021-02-05:
Ah, if you disable `use towers` your problem should be solved. 

[image]

I will close this issue, because there is a solution. Please let me know if it worked!

closed 2021-02-05 by fvrmr

3dparade, 2021-02-05:
Perfect, thanks for prompt help!