Ultimaker/Cura #9244: Benchy hull line fix to get better print surfaces

PuddingBaer91, 2021-02-05:
The benchy hull line problem is discribed by the prusa team and under "development". Please take a look:
https://help.prusa3d.com/en/article/the-benchy-hull-line_124745

I got the idea to let the cura developers implement some test options to get rid of this behaviour. As in the comments mentioned it could be a fix to check before and after layer time. If the layer time differs a lot the print speed should be adjusted so the layer time is not that big. Something like a gradual layer time decrease.
(Please take a look at the article and the comments)

Print a  20*20 mm cube and slice with 10 top and 10 bottom layers and about 15% infill. With 0.1 layer height you can measure 1mm bands at the bottom and the top. The same thing that happens at the benchy's hull where we are going from infill to solid fill.

I hope i could win you to get rid of this problem ;)

Thank you very much.

fvrmr, 2021-02-08:
Hi @PuddingBaer91 thank you. 
We are also aware of this issue. And we would love to have a solution for this. 

Would love to hear possible solutions here over this problem.

Ghostkeeper, 2021-02-08:
The issue is that there's a completely solid layer of skin there. The material normally shrinks when cooling, but due to the extra material there it is unable to shrink. This makes it a bit wider.

Reducing the Skin Expand Distance to 0.5mm (down from 0.95) removed the banding issue because the skin is then no longer blocking the wall from shrinking. There's a bit of infill in between then.

[image]

Is that not an acceptable solution?

PuddingBaer91, 2021-02-08:
@Ghostkeeper never known this solution :) thank you i will try

closed 2021-02-23 by no-response[bot]

appoli13, 2021-07-07:
@Ghostkeeper thank you very much for posting this solution!

I’ll give Cura another shot - I’ve bought & have been using Simplify3D (just got a printer around a year ago) simply because it’s the only slicer that doesn’t have this issue. 
Fingers crossed - it will be nice to use a slicer that has been updated this decade/less than 3 years ago! Haha…

Ghostkeeper, 2021-07-14:
After you slice, be sure to check layer view. The point is that the skin layer at that height is connecting with the wall and shouldn't connect. If it doesn't connect with the wall, there is no difference in the wall at that height from other heights, and as a result you get a smooth wall.

ellisgl, 2021-09-15:
@Ghostkeeper Yeah - Mine was set to 1.25. I slowly decreased it to 0.5, only very minor changes were seen, nothing like what you were able to achieve. Maybe I should do 0.25.

robhaswell, 2022-01-11:
> Reducing the Skin Expand Distance to 0.5mm (down from 0.95) removed the banding issue because the skin is then no longer blocking the wall from shrinking. There's a bit of infill in between then.

Hi, could you repeat this test please? It's possible that environmental differences between the tests might have brought about this apparent solution.

Ghostkeeper, 2022-01-18:
I have printed Benchies a bunch of times in my time at Ultimaker and know that the hull line is a common issue. Adjusting the Skin Expand Distance consistently fixes it for me.

There's a useful theoretical approach when you encounter banding like this. Banding is essentially a sudden change between layers. Sudden changes don't just occur arbitrarily. Something triggered that change. So when you see banding, always look for what is different between the two layers where the banding starts. In this case it's quite obvious: Something is pushing against the wall from the inside.