Ultimaker/Cura #9246: Fuzzy Top

DanielHowden, 2021-02-06:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I do love the fuzzy skin feature and I would love to have that on the top of the part I am printing

**Describe alternatives you've considered**
I have experimented with textures on the model but I doesn't really work for me

**Affected users and/or printers**
For the users who like the fuzzy skin this will be great

**Additional context**
(Add any other context or screenshots about the feature request here.)

fvrmr, 2021-02-08:
Hi @DanielHowden thank you for your feature request. I will discuss this with the team. Keep you posted!

fvrmr, 2021-02-09:
I discussed it with the team, but I will defer this issue because it is not one of our priorities right now. And it would also be very complex to implement. 
Hope you understand!

DanielHowden, 2023-07-02:
I still feel that this would be a good feature.  The part that has fuzzy skin currently has a flat top surface.  It would be great if the top was also fuzzy.