Ultimaker/Cura #9248: Not working purge tower on Geeetech A20M

fespafer, 2021-02-06:
**Application version**
4.8.0

**Platform**
WINDOWS 10 PRO VERSION

**Printer**
GEEETECH A20M

**Reproduction steps**
1. I added printer setup from marketplace in order to activate "shared heater" (which A20m has), since it was removed after 4.6 version
2. Tried to print using 2 colors, but purge tower doesn't work propertly

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
https://ibb.co/QPrXKQR  <-- how the tower is built on first layers who has only 1 color

https://ibb.co/THM3y7C <-- tower is purging now 2 colors in a layer with 2 colors, but there is not base on the second color, so the filament is throwed inside the hollow of the tower and it's sticked to the nozzle

**Actual results**
The filament keep sticked to the nozzel since it has not base on the purge tower and it breaks the structure of the printed object

**Expected results**
The tower must have the same inner diameter all along the printing process, so when 2 colors are printed, there is a base for purging the 2nd color

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-08:
Hi @fespafer thank you.
This is expected behaviour of Cura. Because your second extruder isn't active in the first layers of your model. So it won't print on the prime tower. Because it takes a lot of time while printer.
But you could enable `wipe inactive nozzle on prime tower`. You can read more about this setting here: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/dual/prime_tower_wipe_enabled.md
Please let me know if that works for you!

fespafer, 2021-02-08:
Hi @fvrmr, thanks for your reply. I've already ticked this options, but the result is the same. The tower is growing thinner than expected, so when 2nd colour is needed, there is not base for it and the filament is sticked to the hot end, ruining the model.

fvrmr, 2021-02-08:
Hmm okay, could you share your project file?

fespafer, 2021-02-08:
Sure! 

[SOPORTE SERGIO FINAL.zip](https://github.com/Ultimaker/Cura/files/5943463/SOPORTE.SERGIO.FINAL.zip)

fespafer, 2021-02-08:
Please, check how tower is thinner than needed until layer 43, where 2nd colour starts mixing with 1st colour. thanks in advance

fvrmr, 2021-02-09:
It is still expected behaviour. On layer 43 the second extruders becomes active. And that's why the prime tower will grow bigger. 
I changed colours of the materials to make it more clear. Black is extruder 1 and yellow is extruder 2.

**Layer 19:**
[image]

**Layer 43:**
[image]

We never have experienced any problems with this. 
But I understand that this is a problem for you. A work around to solve this, is adding a little cube that will be printed with the second extruder. So the prime tower will print also with extruder 2 because it is active. As you can see on layer 19 the prime tower is also printing with extruder 2. 
[image]

Hope that this makes sense to you. I will close this issue because it is not a bug. 
But please reach out if you need more help or have questions and off course if this will work with your print!

closed 2021-02-09 by fvrmr

fespafer, 2021-02-09:
Ok, thank you, I see what you mean, but it's a waste of material. Thank you anyway. Regards

fvrmr, 2021-02-09:
Yes I know, but changing this is not one of our priorities, because we never had real problems with it.

Ghostkeeper, 2021-02-15:
You can make the extra object arbitrarily small to save time/material. As long as Cura thinks the material is used on that layer, it'll plan a prime in the prime tower.

fespafer, 2021-02-15:
Great, thank you