Ultimaker/Cura #9249: Hole Horizontal Expansion

alexmatveev1, 2021-02-06:
When specifying the horizontal expansion of holes, the 35th layer in the attached project file is not filled.
Cura version 4.6.1 and 4.8
[Hole.zip](https://github.com/Ultimaker/Cura/files/5937273/Hole.zip)

fvrmr, 2021-02-08:
Hi @alexmatveev1,
I tried to reproduce this in 4.8, but I'm not having the same issue.

[image]

Does this happen with only this file?

alexmatveev1, 2021-02-08:
In Сura v 4.8.0 I tested another 3d model with the same issues.

[Hole2.zip](https://github.com/Ultimaker/Cura/files/5943511/Hole2.zip)

fvrmr, 2021-02-08:
Thank you I can reproduce this with your new file. I will discuss this with the team.

nallath, 2021-02-09:
It's an issue with the Arachne build as well.

[image]

rburema, 2021-02-09:
@nallath Can you put it back on the board then, for additional discussion/decision tomorrow morning?

smartavionics, 2021-02-09:
The following has not been extensively tested but appears to fix the issue...
```
diff --git a/src/layerPart.cpp b/src/layerPart.cpp
index 4084d0ec..2aeaf3d3 100644
--- a/src/layerPart.cpp
+++ b/src/layerPart.cpp
@@ -61,7 +61,7 @@ void createLayerWithParts(const Settings& settings, SliceLayer& storageLayer, Sl
         if (hole_offset != 0)
         {
             // holes are to be expanded or shrunk
-            PolygonsPart outline;
+            Polygons outline;
             Polygons holes;
             for (const PolygonRef poly : result[i])
             {
@@ -74,12 +74,7 @@ void createLayerWithParts(const Settings& settings, SliceLayer& storageLayer, Sl
                     holes.add(poly.offset(hole_offset));
                 }
             }
-            for (PolygonRef hole : holes.unionPolygons().intersection(outline))
-            {
-                hole.reverse();
-                outline.add(hole);
-            }
-            storageLayer.parts[i].outline = outline;
+            storageLayer.parts[i].outline.add(outline.difference(holes));
         }
         else
         {
```

smartavionics, 2021-02-09:
Yeah, that wasn't quite right, you still need to union the holes, so this is a better fix...
```
diff --git a/src/layerPart.cpp b/src/layerPart.cpp
index 4084d0ec..e3d4e397 100644
--- a/src/layerPart.cpp
+++ b/src/layerPart.cpp
@@ -61,7 +61,7 @@ void createLayerWithParts(const Settings& settings, SliceLayer& storageLayer, Sl
         if (hole_offset != 0)
         {
             // holes are to be expanded or shrunk
-            PolygonsPart outline;
+            Polygons outline;
             Polygons holes;
             for (const PolygonRef poly : result[i])
             {
@@ -74,12 +74,7 @@ void createLayerWithParts(const Settings& settings, SliceLayer& storageLayer, Sl
                     holes.add(poly.offset(hole_offset));
                 }
             }
-            for (PolygonRef hole : holes.unionPolygons().intersection(outline))
-            {
-                hole.reverse();
-                outline.add(hole);
-            }
-            storageLayer.parts[i].outline = outline;
+            storageLayer.parts[i].outline.add(outline.difference(holes.unionPolygons()));
         }
         else
         {
```

closed 2021-02-12 by nallath