Ultimaker/Cura #9251: Zooming into model which disappears off the bottom of the window.

Peter2055, 2021-02-07:
Version 4.8.0

The issue has existed as long as Ive been using Cura when zooming into the model it will often disappear off the bottom of the screen so its no longer visible at all. Could the build plate be positioned in the window so the model doesn't disappear when zoomed in? Though the view can be reset its always zoomed out. When zooming back in it disappears again. Ive just accepted the issue in the past and simply gave up trying to see the parts of the model that can only be seen zoomed out. If the build plate was repositioned in the window towards the vertical middle it might help? Ive tried to re orient the application window to a 4x3 aspect but it doesn't help. Im using a Mac Book air 13. I was wondering if it will display better on a native 4x3 aspect monitor?

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

StuSerious, 2021-02-08:
You could've put a bit more effort into your report.. 
Could you provide screenshots or a small clip of the issue? Or maybe a project file?
Does turning on "Center camera when item is selected" in 'Preferences > Configure Cura...' help with the issue?
It could also be a driver on your side acting up, I think logs could help to check for OpenGL errors!

fvrmr, 2021-02-08:
Hi @Peter2055 this depends on the height of your model. 
But I will discuss this with the team. Keep you posted.

A good work around is the one @StuSerious is suggesting.

GregValiant, 2021-02-09:
In Orthogonal view, the setting "Zoom toward mouse direction" doesn't work.  But after zooming I can click the center scroll button and drag the view back to the center and then zoom in some more.  Using Perspective view "Zoom toward Mouse" works as it should and you can move the mouse to direct the model to stay in the center of the screen during a zoom.

closed 2021-02-15 by Ghostkeeper

Ghostkeeper, 2021-02-15:
According to our anonymous usage statistics, most scenes that people print are fairly small, 10-50mm. We've decided to lower the default camera position such that the default camera origin is now 30mm above the build plate rather than 100mm. If you load something small in, it'll still go over the top of your model if you zoom in far enough, but it should be better for most prints now.

Myself I can't really live without checking that preference option to move the camera to the selected object. However this was a decision from UX that I can't overrule. The rationale is that it's confusing/annoying if the camera moves automatically without the user's control.

commit 0c9175d7 (2021-02-15): Lower camera default position by 70mm

The previous default position was aimed at 100mm above the build plate. This is taller than most models that people load into Cura. It's now aimed at 30mm. Models up to 50mm are way more common to print, so this brings it right in the butter zone for most people, hopefully.

Fixes #9251.