Ultimaker/Cura #9252: [4.8]Intermittent freezing Cura

userosos, 2021-02-07:
**Application version**
4.8

**Platform**
Windows 10 64 bit 20H2 19042.746

**Printer**
Custom

**Reproduction steps**
1. Move an parts or just do nothing

**Actual results**
Sometimes i can see an freezing (~5-10 sec) an window Cura

[cura.log](https://github.com/Ultimaker/Cura/files/5939215/cura.log)

fvrmr, 2021-02-08:
Hi @userosos thank you for your report. 
I looked into your log file but couldn't find anything.

Could you log out in Cura and see if that helps from freezing.

userosos, 2021-02-08:
> 
> 
> Hi @userosos thank you for your report.
> I looked into your log file but couldn't find anything.
> 
> Could you log out in Cura and see if that helps from freezing.

Only the Cura window hangs. The system continues to work fine.

fvrmr, 2021-02-08:
> Or did you mean logging out of your Cura account?

Yes I mean your Cura account!

userosos, 2021-02-08:
> 
> 
> > Or did you mean logging out of your Cura account?
> 
> Yes I mean your Cura account!

I unlogged and just moved the part. It not helped.
[cura.log](https://github.com/Ultimaker/Cura/files/5943968/cura.log)

Ghostkeeper, 2021-02-09:
I'm using the exact same laptop as you, so the hardware is not the problem.

fvrmr, 2021-02-09:
I still couldn't find anything weird in your log file. So unfortunately I can't help you with this right now, because we can't reproduce this and can't find the problem. 
So we'll have to wait for others with this problem.

userosos, 2021-02-09:
> 
> 
> I still couldn't find anything weird in your log file. So unfortunately I can't help you with this right now, because we can't reproduce this and can't find the problem.
> So we'll have to wait for others with this problem.

Ok. I think that ii some problem with an other software maybe.

userosos, 2021-02-13:
Who can tell me how to diagnose the problem of frequent Cura hangs from third-party software/settings?

Ghostkeeper, 2021-02-15:
If you suspect that other software is causing Cura to hang, the obvious way to test that would be to disable or shut off that software and see if the issue is resolved then.

Perhaps it helps to narrow down when exactly Cura freezes. Maybe it freezes after 10 seconds of inactivity? Maybe it freezes when you're starting a slice?

userosos, 2021-02-15:
> If you suspect that other software is causing Cura to hang, the obvious way to test that would be to disable or shut off that software and see if the issue is resolved then.
>
> Perhaps it helps to [...]

I noticed that it hangs every ~30-60 seconds.

userosos, 2021-02-18:
I think it happens every 5-10 mouse clicks.

Ghostkeeper, 2021-02-19:
Maybe it is the autosave? The symptoms are like that.

userosos, 2021-02-19:
> Maybe it is the autosave? The symptoms are like that.

The autosave in Cura?

fieldOfView, 2021-02-19:
Autosave saves preferences and setting changes to your configuration folder.

userosos, 2021-02-19:
> Autosave saves preferences and setting changes to your configuration folder.

In the properties of the folder did not find autosaves.

closed 2024-12-08 by github-actions[bot]