Ultimaker/Cura #9254: Support Blocker Doesnt work if under 5mm wide

PoiMechanic, 2021-02-09:
**Application version**
4.7 and 4.8

**Platform**
Macbook Pro 16" OS Catalina, AMD Radeon Pro 5600M

**Printer**
Prusa MK3S

**Reproduction steps**
Model requires supports up the column in the middle. If I have support walls set to anything above 0 (2 in my case) the supports will close in with no openings so I have to cut them off (that might be a bug by itself).  The main bug is when I use a support blocker or an imported object that is set to "Don't Support Overhangs" to create a gap in the supports, it won't work if the blocker is under 5mm wide. I need 1 mm wide so the overhand has minimum distance without support.

**Actual results**
Nothing happens unless I set the blocker to 5mm.

**Expected results**
Any size blocker should work. 

[image]
[image]
[image]

**Project file**
I tried to upload both a STL and a 3MF and I get the message "We don't support that file type" so I'm not sure how else to upload it. 

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5948665/cura.log)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-09:
Hi @PoiMechanic try to change the `support join distance` to 0.  That should probably fix your problem.
Please let me know if that works for you!

PoiMechanic, 2021-02-09:
Unfortunately that didn't so anything. Tried just doing that by itself and with the support blockers.

fvrmr, 2021-02-09:
I tried it myself, and here is my support blocker 4 mm. 
[image]

Could you share your project file? You can share is as a .zip

PoiMechanic, 2021-02-09:
Hmm thats weird. Looks like your Support Wall Line Count is at 0 but I tried that too and it didn't work. I attached a zip of the file. Profile is set to Normal with nothing changed except for the Support Wall Line Count is set to 1
[Cura Issue.3mf.zip](https://github.com/Ultimaker/Cura/files/5954261/Cura.Issue.3mf.zip)

fvrmr, 2021-02-11:
Hi @PoiMechanic thank you for your project file. 
It is working for me. The `support join distance` is 0 and `support wall line count` is 1. 

[image]

Here is the file from the screenshot above. Can you take a look at it and see if it is still happening?
[PI3MK3M_Cura Issue.zip](https://github.com/Ultimaker/Cura/files/5963953/PI3MK3M_Cura.Issue.zip)

And does this happen on other models? If so could you share one of those.

PoiMechanic, 2021-02-11:
Hot diggity damn that is the fix! I guess the support join distance is normally set to 2 so that must be the issue. Thanks so much @fvrmr

fvrmr, 2021-02-11:
Your welcome! 
I will close this issue, because it is solved!

closed 2021-02-11 by fvrmr