Ultimaker/Cura #9256: [at least 4.8] Automatically selected models do not center the camera while auto-center is enabled.

StuSerious, 2021-02-09:
**Application version**
Cura 4.8 | Windows

**Platform**
**OS:** Windows 10 Home
**CPU:** R5 2500U w/ Vega 8 GPU
_Note: this is the platform I discovered the "bug" on, but I can reliably reproduce it in every other platform I tried._

**Printer**
Ultimaker S5
_Note: I can also reproduce the "issue" on every custom profile I have!_

**Reproduction steps:**
1. Head to 'Preferences > Configure Cura...';
2. Under "**Viewport behavior**", enable "Center camera when item is selected";
3. Under "**Opening and saving files**", enable "Select models when loaded";

**Screenshot(s)**
adding screenshot after report is published to rename attachment with issue number

**Actual results**
If both "Select models when loaded" && "Center camera when item is selected" are active, when a new model is loaded it is automatically selected but it does not trigger the automatic centering of the viewport / camera!

**Expected results**
If both "Select models when loaded" && "Center camera when item is selected" are active, when a new model is loaded it is automatically selected and the viewport is centered onto it!

**Project file**
I don't think it's needed, but I can provide it if necessary!

**Log file**
I don't think it's needed, but I can provide it if necessary!

**Additional information**
I know this is an extremely low priority issue and may very well be expected behavior, but every time I load a model my brain kinda expects Cura to center it while both of these are enabled!
Again, this is just quality of life fix and would only take a click off of my workflow, so I don't really care that much since it has become second nature to just click off the model and reselect it to center it perfectly like my OCD so urgently demands.
I was just wondering if this is a 15 minute fix worth doing or intended behaviour and it's flying over my head :D

Oh and as always thanks to everyone on the team, I hope I don't loose you time opening unimportant issues!

StuSerious, 2021-02-09:
Uh also, I forgot, I'm open for suggetions for a shorter title 🤣

fvrmr, 2021-02-09:
Hi @StuSerious thanks for your report. 
I will discuss it with the team. Keep you posted!

fvrmr, 2021-02-11:
Hi @StuSerious, 
I have discussed it with the team. And I have to defer this issue. 
It is not a big issue and priority for us and with the work around to click on the model when loaded it will zoom in to your model. But it takes an extra click. 😉 

Hope you understand!

StuSerious, 2021-02-12:
Thanks alot @fvrmr!
I can totally see why you deferred, no big deal with the workaround!
As always thanks a lot for your time!

Should I close the issue?

fvrmr, 2021-02-12:
> Thanks alot @fvrmr!

Your welcome

> Should I close the issue?

No you can leave it open, because it is deferred and not rejected 😉

closed 2023-06-13 by github-actions[bot]