Ultimaker/Cura #9257: Slicer bug

drewlucas98, 2021-02-09:
**Application version**
4.8.0
**Platform**
Windows 10 pro 64 bit with an rx5700xt

**Printer**
ender 3v2

**Reproduction steps**
1. try to print this https://www.thingiverse.com/thing:3612920
2. rotated object so that its flat side is facing down. 
3. sliced
4. top of the mount has gaping holes in the preview

**Screenshot(s)**
[image]
[image]

**Actual results**
same thing

**Expected results**
obviously it wouldn't have those holes in the preview

**Project file**
https://drive.google.com/file/d/1P3ROjtgEneVnVkWH4EbPZW9rrNXujGlh/view?usp=sharing

**Log file**
too hard dont have time for that

**Additional information**

StuSerious, 2021-02-09:
@drewlucas98 I can't reproduce the issue on my Ender-3, but I would bet thatwhat you are seeing is the effect of excessive coasting!
Try to turn off '**Coasting**' or to reduce its volume [you can find the setting tree under the Experimental category].

If after turning off coasting it still has those gaps, can you please share your Project File? 
Thanks!

fvrmr, 2021-02-09:
Hi @drewlucas98 I can't reproduce this either. This is how I slice it on a Ender 3. 

[image]

A project file would be helpful!
Disabling coasting is a good suggestion! Thanks @StuSerious

closed 2021-02-23 by no-response[bot]