Ultimaker/Cura #9258: Cooling over infill

Julius971, 2021-02-09:
My idea builds on the Lift Head functionality to wait for the desired print time. I noticed stringing/oozing caused by this.

**Describe the solution you'd like**
The idea is, not to locate the pause position outside the print (so the oozed material will harm the skin), but to place it on the innermost infill part of the current layer. (or some infill close). The nozzle should be kept over an empty area of the infill, so the material leaving the nozzle gets trapped there. After the delay time the nozzle wipes over the next infill line (optional) and continues the next layer.
This should take care of any stringing on the outside. Maybe an extra prime amount after delay moves should be considered. 
Or (this might be a completely new request) variable extra prime amount like mm³ per seconds traveled/paused.

**Affected users and/or printers**
FDM

fvrmr, 2021-02-11:
Hi @Julius971 thank you for your feature request. 
I will discuss this with the team. Keep you posted!

fvrmr, 2021-02-12:
This feature request is on our backlog.
Developers see CURA-8022