Ultimaker/Cura #9260: Bedleveling options.

Julius971, 2021-02-10:
**Describe the solution you'd like**
Really simple idea, but it could improve first layers alot.
There should be a new placeholder for the start gcode where cura pastes a bed leveling command. And for the command there should be settings wherr you can set the bed leveling boundaries to fit your print. This would use the configured amount of bed level points only on the used printsurface to give a higher resolution bedleveling for the important area. 
This should be in cura, because it could calculate the best area possible. (Could add options like "exclude brim/skirt")

fvrmr, 2021-02-11:
Hi @Julius971 thank you again for your feature request. 
Unfortunately I have to reject this feature request. Because it very printer specific. 
A work around is to add this manual to your start g-code. 

Hope you understand!

closed 2021-02-11 by fvrmr

Julius971, 2021-02-11:
Or could it be possible to add placeholders like marginleft/right/front/back to implement them in the start gcode. But calculating the values by hand is a bit tricky. Done by cura it would simply looking for max and min values of the first layer and use them.
minx = marginleft
maxx = marginright etc