Ultimaker/Cura #9261: Electricity cost

Julius971, 2021-02-10:
Quick and easy feature. Machine settings could have an option like average wattage. And you could add electricity cost €$£¥ per kwh.(please tell me as a non native english speaker how to say that😂)

StuSerious, 2021-02-10:
+1! This would be cool! 
In the meantime if you want, a cool resource is [this spreadsheet](https://github.com/CNCKitchen/3D-Printing-costs) from Stephan over at CNCKitchen.
I use a modded version of this to run my numbers!

fvrmr, 2021-02-11:
Hi @Julius971 thank you for your feature request. I will discuss this with the team. Keep you posted!

Liger0, 2021-02-19:
That would be useful specifying wattage on the printer profile and a global setting with your KWh

firefighter331, 2021-02-20:
That may even be something that could be made specific to Ultimaker printers by internally measuring the power usage and reporting it to the user via Cura. The user would obviously need to define the cost of electricity, but technology can measure it and do the calculations based on that.

I just don't see it being very accurate unless there's something actively monitoring power usage. There are a lot of variables and it would be hard for devs to have the software accurately _guess_ the power usage. The motors aren't constantly moving. Even when they are, they're not all moving at once. Even when they're all moving at once, they're not all moving at the same speed, so they draw a different amount of power. Plus you have different heating cartridges and bed heaters that draw different amounts of power and have different efficiency levels. One print bed may be insulated better, making it more energy efficient, but Cura doesn't know that.
I don't see it being remotely accurate with all those variables, **UNLESS**, like I said, there is some type of meter that's actively watching the power usage and keeping count.

StuSerious, 2021-02-23:
> I just don't see it being very accurate unless there's something actively monitoring power usage. There are a lot of variables and it would be hard for devs to have the software accurately guess the power usage. The motors aren't constantly moving. Even when they are, they're not all moving at once. Even when they're all moving at once, they're not all moving at the same speed, so they draw a different amount of power. Plus you have different heating cartridges and bed heaters that draw different amounts of power and have different efficiency levels. One print bed may be insulated better, making it more energy efficient, but Cura doesn't know that. If the printer is running on USB rather than a card or flash drive, then there's also power being drawn by the computer it's connected to.

I think this is just overcomplicating a simple problem: have a rough idea of how much a print is going to cost you in the power bills.

I don't know about the other parts of the world, but in the EU (where Ultimaker is from) all machines by spec have to have an average power consumption metric (you can read more about it [here](https://europa.eu/youreurope/business/product-requirements/labels-markings/energy-labels/index_en.htm)), so you can just read the label on your machine and plug in that number in a simple calculator to get a farely accurate idea of what it's going to draw during standard operation. 

I'd imagine most people don't care about knowing consumption down to the last Joule, just an average figure we can use to estimate power draw and therefore cost.

Ghostkeeper, 2021-03-01:
I don't think the motors are really going to be the problem on average, but the build plate definitely is. Heating the build plate to 60 degrees for PLA vs. 80 degrees for ABS is a huge (~50%) difference in power draw of the build plate. And the build plate, at least for Ultimaker printers, is the majority of the power draw of the entire system.

As for updates from the Cura team on this issue. The UX designer wants to discuss this with the UX designers of other software products of Ultimaker (the printer management system, specifically). This could take a while, but we'll keep you posted if something is new.

I wouldn't be opposed to add a field for average electricity cost per hour, show that in the print statistics and have it be done with that.

If the electricity cost is specified in watts (or other units of energy per time) then we'd also need a field for the cost of those joules, since that's the only way we can then collate that with the material costs for the actual useful metric of total part cost. Instead, it might be easier to just have one field for electricity cost per hour (like in $/h or whichever currency the user has). And I think Ultimaker would then also look into the cost of maintenance and labour, seeing as how they like to focus on business-to-business nowadays.

3Duke3, 2021-07-03:
Alright, I submitted a request #10058 which has been closed due to being deemed similar to this one.

I thought it may be best to have a proper discussion around the print costs displayed in Cura, but more importantly how and what you guys and others use the displayed costs for, so that we can find a proper real world purpose + solution to a problem.

I like the idea of adding electricity costs into the print cost, but if you are using the pricing for any kind of serious financials, I think electricity costs are probably the most insignificant portion of a prints price makeup. ie, if the difference of $.15c - $.45c in power consumption is affecting whatever you use the displayed price for, then we may as well build in 100+ other fields for things such as, machine cost over usable lifetime, maintenance and repair costs, insurance, wages for person preparing and cleaning models, the list could go on for ever. What I am trying to say is that displaying electricity costs seems like a fancy feature with no meaningful benefit to anyone, however I believe their may be a better solution...

I would like to propose a change that would require only a few additional fields that I think could solve a wider scope of pricing applications.

The idea to would be to keep the existing cost of filament field
+ add a filament sell price field (in which a user can mark up the spool of filament)
+ add a machine running cost as a per hour value (this would be specific to each printer)
+ add a machine selling cost, as a per hour value (also specific to each machine)

the end result would be that after slicing each print you could see what that print will cost (cost of filament amount used + (machine running cost * print duration)). As well as a selling cost if desired (marked up amount of filament used + (machine (selling cost * print duration))

The machine running/selling costs can be determined by the user. Rather than having 100 fields for all aspects of printer running costs, the user simply averages out their own requirements and determines their own averaged hourly rate.

So what and how do you use the currently displayed print cost for?

3Duke3, 2021-07-03:
Me personally, I print things for people and invoice them to make a profit. So in an ideal world, I would love it if Cura could show me what it costs to print an item, as well as a calculated sale price.

I realize there are all sorts of spreadsheets I can use where I take the filament weight and print time ect, input into spreadsheet and generate a price. But when your printing /quoting a reasonable amount of parts each and every day, inputting all this information into a spreadsheet is hugely time consuming.

I would love to drop a part into Cura, slice it with a basic/generic profile and boom see that it costs x amount of dollars, and Y amount to invoice. Even if it only showed me the cost that would be tremendous, asIi can tweak the imputed $ values to accommodate a profit margin, or easily calculate a markup the shown cost.

Keen to hear peoples thoughts, as I think a solid costing function in Cura would be a huge benefit to Ultimaker business customers especially if their accounts departments want to see better cost breakdowns, or even if they are internally invoicing produced parts to other departments. @Ghostkeeper?

Ghostkeeper, 2021-07-12:
I think a lot of these costs are better tracked in different software. Cura is not meant to be the accounting system of your business. However, the power of putting it in Cura lies in that the user can adjust the print parameters - without printing it - and get a prediction of the cost, so that they can minimise those costs by adjusting print parameters.

Cura can only predict costs from what it knows: Material usage, printing time and the settings used. You could *give* Cura all the rest of the information, like how much time you're spending on post-processing and administration and whatnot, but that doesn't play into the advantage that Cura has over other software, so I don't think it should be done in Cura.

Electricity cost, or a cost of operating the printer per hour, would directly play into this advantage so I think that is something we could add.

I'll ask the UX designer if anything came out of his discussion with the other UX designers yet.

3Duke3, 2021-07-12:
> However, the power of putting it in Cura lies in that the user can adjust the print parameters - without printing it - and get a prediction of the cost, so that they can minimise those costs by adjusting print parameters.

Totally agree with you. I think if we could have a $value field per print hour this would add a lot of benefit to many users (especially me... like it would literally save me +5 hours each week being able to loosely see a time based $ value, so I can more accurately quote direct from Cura as I am inspecting the files). I think it should be relatively easy to implement also.

toozie21, 2022-05-14:
I was just googling to see if a plugin to do this already exists and stumbled across this.  For me, something as simple as putting in the average cost of kWhr for my locale and wattage of the machine reported (which I know is a max value, that is fine), and then letting Cura multiply by the hour estimate is good enough for me.  I am looking for a rough approximation, so I don't need anything too complicated.

InstantMuffin, 2023-07-15:
I would appreciate this feature as well. I'd be fine with a rough estimate. -+20% accuracy would be enough for me.
Anything other than rough estimates would need either metering outside of the printer, inside (a current sensing circuit in series with the psu output and the power input of the mainboard, with a clamping circuit connecting the vdrop to the adc), or more elaborate logic using the printer's firmware (using the power rail's voltage and then extrapolating things from stepper current and usage, and cartridge, heatbed, and fan pwm with preentered current usage).

distante, 2023-09-17:
I would be happy with a hard value that I can put per hour, to be honest.

amcolash, 2023-10-16:
I would love this cost baked into Cura as well. In the meantime, I did find this plugin (search power cost) which does exactly what I want! It shows a little power estimate near the filament cost estimate when printing. It isn't the nicest having two costs instead of a single summed total, but better than nothing for now 😁 

https://github.com/YiamiYo/Cura-PowerCostPlugin

[image]