Ultimaker/Cura #9262: Cura adds a new layer in my hollow object

nay0219, 2021-02-10:
**Application version**
4.8.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
Ultimaker S5

**Reproduction steps**
1. Uploaded STL file made from C4D. Printed with PVA.
2. Sliced--it's empty when "Preparing" but once sliced, the "Preview" shows a layer formed on the base of the design.

**Screenshot(s)**
[image]
^^what i want it to look/print  (PREPARE stage). It's a lampshade so it should be hollow inside.

[image]
^^^ what it looks like after sliced. As you can see there is a yellow layer on the bottom that wasn't visible in the PREPARE stage.

[image]
^^^how it actually printed. Thought that maybe the issue was the brim settings. But if I turn the Brim off the layer still appears. It also is a pretty thick layer.

[image]
^^^settings

**Actual results**
The lampshade prints with a layer on the wider end of the sahde. Thought that maybe it was a brim issue and that I can maybe pop it off but once printed, there's a thick layer that cannot be cut/popped off. Even if I rotate the object 180 degrees on CURA the layer still shows up in the same location. It may be a model issue? 

**Expected results**
Should be hollow throughout. There should be no layer in between.

fvrmr, 2021-02-11:
Hi @nay0219 could you share your project file? So I can have a look at your model myself!

nay0219, 2021-02-11:
> Hi @nay0219 could you share your project file? So I can have a look at your model myself!

[lamp low poly.zip](https://github.com/Ultimaker/Cura/files/5965998/lamp.low.poly.zip)

GregValiant, 2021-02-12:
Cura didn't report any errors when opening the file but it doesn't typically make things up.  I think there were reversed faces 7.2mm tall at the bottom.  Service.Netfabb.com repair utility fixed them.  The attached is the repaired model from the STL so it isn't scaled the same as the 3mf was.
[lamp_low poly_fixed.zip](https://github.com/Ultimaker/Cura/files/5971849/lamp_low.poly_fixed.zip)

fvrmr, 2021-02-16:
Hi @nay0219 could you let us know if the fix of @GregValiant works for you?

closed 2021-03-02 by no-response[bot]