Ultimaker/Cura #9263: Odd behavior with Make Overhangs Printable

KenLucke, 2021-02-10:
**Application version**
4.8

**Platform**
Mac OS
[image]

**Printer**
Creality CR-10s (with settings to make CR-10v2)

**Reproduction steps**

1. Loaded model
2. Set slicer settings, including Make Overhangs Printable
3. Sliced

**Screenshot(s)**
Top:  No "Make Overhangs Printable"
Middle: With "Make Overhangs Printable
Bottom: Raw Model
[image]

**Actual results**
Cura adds a center portion of the body that does not exist

**Expected results**
Shouldn't do that

**Project file**
[Gothic Lantern top.3mf.zip](https://github.com/Ultimaker/Cura/files/5961258/Gothic.Lantern.top.3mf.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5961298/cura.log)

**Additional information**
(Extra information relevant to the issue.)

nallath, 2021-02-11:
`Cura adds a center portion of the body that does not exist`

That is exactly what the make overhangs printable setting is supposed to do.

fvrmr, 2021-02-11:
> `Cura adds a center portion of the body that does not exist`
> 
> That is exactly what the make overhangs printable setting is supposed to do.

I will close this issue because it is expected behaviour from Cura. 
Hope you understand. Please let me know if you have questions.

closed 2021-02-11 by fvrmr