Ultimaker/Cura #9264: Cura Marktplatz

MeckyGT, 2021-02-11:
Hallo,
At the cura Marketplace    Calibration Shapes     the Link in the left corner isn´t right.
And the Version is to day 1.0.9     not 1.07.
The change is very interesting because the is now the PETG Test Tower.
Thank you.

nallath, 2021-02-11:
The link is something that @5axes needs to fix. The newer version of the plugin is still under review by us, but you can expect that to be released soonish.

closed 2021-02-11 by nallath

5axes, 2021-02-11:
What is this link you're talking about?

nallath, 2021-02-11:
@5axes I think he means this (there is a double https:// in there):

[image]

MeckyGT, 2021-02-11:
Yes that is what I meen.
--
Gesendet über myMail für Android Donnerstag,