Ultimaker/Cura #9265: Tree support assertion failure / crash

BagelOrb, 2021-02-11:
**Application version**
current master / the UMH-... branch

doesn't reproduce on 4.8

**Platform**
WSL

**Printer**
UM3

**Reproduction steps**
1. load 
[UM3_cube_cm_bug.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/5963985/UM3_cube_cm_bug.3mf.renamed.zip)

**Screenshot(s)**
[image]

**Actual results**
```
CuraEngine: /mnt/c/Development/CuraApp/CuraEngine/src/TreeSupport.cpp:684: const cura::Polygons& cura::ModelVolumes::calculateCollision(const RadiusLayerPair&) const: Assertion `ret.second' failed.
```

**Expected results**
No tree support, no crash

**Additional information**
Unlikely this is related to the fact that I'm on a feature branch, because that branch is concerned with infill only.

fvrmr, 2021-02-11:
Hi @BagelOrb thanks for your report. Will discuss it with the team!

Ghostkeeper, 2021-02-11:
I'm not getting a crash with that project file on master.

It produces no infill since the infill pattern is undefined (part of your project which is not in the master branch), but setting that back to Triangles everything looks fine.
[image]

BagelOrb, 2021-02-11:
You might want to take a second look at the code where the assertion failure happens. Otherwise we could just clost this issue as non-reproducible...

fvrmr, 2021-02-12:
Hi @BagelOrb we couldn't reproduce this.
So I will close this issue.

closed 2021-02-12 by fvrmr

Ghostkeeper, 2021-02-18:
The assertion in question tests that the collision volume is not empty on that layer, basically, after having offset the mesh with the collision radius. Perhaps it would be colliding with the model if that collision volume should not be empty?