Ultimaker/Cura #9267: [4.8.0] Bad skirt outline calculation

FFY00, 2021-02-11:
**Application version**
4.8.0

**Platform**
Arch Linux

**Printer**
Ender 3 V2

**Reproduction steps**
1. Load https://www.thingiverse.com/thing:2515926
2. Enable skirt
3. Slice

**Screenshot(s)**
[image]

**Actual results**
The big cylinder piece does not get included in the skirt outline.

**Expected results**
The skirt outline should not conflict with any piece.

**Project file**
[CE3_Solder_holder_56mm_spool.zip](https://github.com/Ultimaker/Cura/files/5967811/CE3_Solder_holder_56mm_spool.zip)

fvrmr, 2021-02-12:
Hi @FFY00 thank you.
It looks like this is a problem with the model. Because if you look at the first layer, you can see the big cylinder is not aligned at the same height as the spool holder and the ring. 
[image]

A simple solution for this is to move your model -0.02 on the z axis. 
[image]

Hope this works for you. 
I will close this issue because it is not a bug with Cura.

Please let me know if you have more questions!

closed 2021-02-12 by fvrmr

FFY00, 2021-02-16:
Hi, thank you for the quick response! I ended up just printing as-is, it worked fine as the gap seems to be pretty small. Would it make sense to have cura warn about objects which are completely in the air?

fvrmr, 2021-02-16:
Great to hear that it did work, it is only one layer so that is mostly not a problem indeed.
Well, that's difficult to tell, because sometimes it is mean to be not to touch the build plate. Also Cura already takes a few mm of the bottom of the object to place it.