Ultimaker/Cura #9271: Cura 4.8 Open File crashes

dacobi, 2021-02-12:
**Application version**
Running Cura 4.8 

**Platform**
Manjaro 20.2.1

**Printer**
Prusa i3

**Reproduction steps**
Open Cura
Click Open File

**Screenshot(s)**
None

**Actual results**
Cura crashes

**Expected results**
Cura should open file 

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5972555/cura.log)

**Additional information**
Console output:

[klejs@KlejsPrime ~]$ cura
/usr/lib/python3.9/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/usr/lib/python3.9/site-packages/UM/Settings/InstanceContainer.py:53: DeprecationWarning: invalid escape sequence \d
  version_regex = re.compile("\nversion ?= ?(\d+)")
/usr/lib/python3.9/site-packages/UM/Settings/InstanceContainer.py:55: DeprecationWarning: invalid escape sequence \w
  type_regex = re.compile("\ntype ?= ?(\w+)")
/usr/lib/python3.9/site-packages/UM/VersionUpgradeManager.py:98: DeprecationWarning: invalid escape sequence \.
  ".*\.lock",       # Don't upgrade the configuration file lock. It's not persistent.
/usr/lib/python3.9/site-packages/cura/UI/ObjectsModel.py:72: DeprecationWarning: invalid escape sequence \(
  self._naming_regex = re.compile("^(.+)\(([0-9]+)\)$")
[MainThread] UM.Application.initialize [144]: Initializing Ultimaker Cura
[MainThread] UM.Application.initialize [145]: App Version 4.8.0
[MainThread] UM.Application.initialize [146]: Api Version 7.4.0
[MainThread] UM.Application.initialize [147]: Build type None
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/share/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/share/cura/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib32/uranium must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/bin/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/Resources/uranium/plugins must be a folder.
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/Resources/cura/plugins must be a folder.
[MainThread] UM.Resources.__initializeStoragePaths [446]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [456]: Config storage path is /home/klejs/.config/cura/4.8
[MainThread] UM.Resources.__initializeStoragePaths [464]: Data storage path is /home/klejs/.local/share/cura/4.8
[MainThread] UM.Resources.__initializeStoragePaths [476]: Cache storage path is /home/klejs/.cache/cura/4.8
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/lib/python3.9/site-packages/UM/../plugins must be a folder.
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [54]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [118]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [126]: Yay, we got at least OpenGL 4.1 core: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [173]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [180]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [184]: Initializing version upgrade manager ...
"/proc/7064/root"
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [165]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [186]: Loaded user packages management file from /home/klejs/.local/share/cura/4.8/packages.json
[MainThread] UM.PackageManager._saveManagementData [259]: Package management file /home/klejs/.local/share/cura/4.8/packages.json was saved
qt.qpa.xcb: QXcbConnection: XCB error: 5 (BadAtom), sequence: 503, resource id: 0, major code: 20 (GetProperty), minor code: 0
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [115]: Loading plugin configuration file '/home/klejs/.config/cura/4.8/plugins.json'
[MainThread] UM.PluginRegistry.addPluginLocation [250]: Plugin location /usr/libx32/cura must be a folder.
[... 701 lines cut ...]
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f188d8f6640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007f1894d99640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x00007f18a7e12740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 847 in run
  File "/usr/bin/cura", line 235 in <module>
Segmentation fault (core dumped)

fvrmr, 2021-02-15:
Hi @dacobi thank you for your report.
You are running python 3.9 and that's why you get a segmentation fault. 
Cura is currently on python 3.5.7.

Try to run the AppImage of Cura. That will probably solve your problem. 

Please let me know if run the AppImage your problem is solved.

dacobi, 2021-02-15:
> Hi @dacobi thank you for your report.
> You are running python 3.9 and that's why you get a segmentation fault.
> Cura is currently on python 3.5.7.
> 
> Try to run the AppImage of Cura. That will probably solve your problem.
> Please let me know if run the AppImage your problem is solved.

Hi and thanks for your reply.

Running as AppImage works perfectly :)

closed 2021-02-15 by dacobi