Ultimaker/Cura #9272: Timer for TFT screen in pause-at-height script

ryanaxiom, 2021-02-12:
I know a previous version of this was marked as closed but I think this would be a useful addition.

In https://github.com/Ultimaker/Cura/issues/8751 Robblot blamed the problem on the TFT screen and I think he is likely correct that the TFT is causing the incorrect pause functionality. With the stock firmwares (TFT and mainboard), Artillery added a manual pause button which prevents normal functioning of any of the pause commands in Cura (pause at height, color change, etc). The pause command works and the printer parks in the pause position but only for about 10-20 seconds and then simply proceeds on with the print.  The TFT does not update with a pause indication on the screen.  The TFT still has the option to push pause which does result in an actual pause but ends up being over the printed object. It may have something to do with the TFT pause button logic since the USB and SD slots are actually included on its board. Perhaps it continues to stream the gcode after ignoring the pause since it has its own pause?

Filament change and all other pause options that I could find existing in the software did not work.  In fact, it seemed the filament change was simply ignored.  When I came back just after where the change command was, it had not ejected the filament and was happily continuing with the print (before I could insert my magnets... :(  Perhaps Artillery feels you should pause the print manually on the TFT and use the TFT to unload/load the filament directly

**Perhaps adding a G4 Sxx delay option would fix the existing functionality of the "pause at height" in CURA for affected printers** since it should just wait for that time to expire. I haven't tried that yet.

There are a couple of workarounds that I've found or thought of.

1) Update TFT and mainboard firmware. See https://www.thingiverse.com/thing:4294049 . I have not done this yet, but plan to do so very soon. Many have had success and digant is maintaining this.  Unfortunately, due to the TFT and mainboard communication setup, it is challenging (or impossible) to update the mainboard firmware without opening the printer.  I will likely do this in the next few days since there are one or two other improvements from the stock firmware that I would like to have.

2) Plane cut and print separate files one after another.  To do this, one must make sure on the later files to include a part (small throwaway cylinder perhaps) that will raise the z level to at least what is required.  This is needed because the printer doesn't directly raise the z height but instead will start from home and go directly to the start of the higher layer possibly hitting and dislodging the previously printed object. Further, you have to do this soon after the first print finished because it does not maintain temperatures (essentially this is like a cold pause).

3) Use the filament runout sensor to cause a hot pause. This is better and can be accomplished as in this video. https://www.youtube.com/watch?v=y_WvqhpclNc&ab_channel=Rich%27s3DArtilleryprinting . The advantage of this is that the pause will be a hot pause so the plate will stay warmed. The disadvantage is that you cannot use the filament runout sensor as intended when doing this and requires writing a new script or editing the gcode manually.

So 1 is the likely the best option but requires firmware updates and likely opening the printer to do so. 2 works without modification, but you have to manually intervene soon to keep the bed hot and make sure the second print is set not to crash into the first (by adding a waste object). 3 will give warning beeps and hot pause but requires some setup and you lose the filament runout sensor for that particular print.

**Affected users and/or printers**
Sidewinder X1 and Genius users. Perhaps other TFT based printers?  Also, anyone who wants a fixed pause length for the "pause at height/layer" script.

ryanaxiom, 2021-02-12:
I haven’t tried printing via usb connection to Cura so I will try that to see if that works. Perhaps it will work that way. Printing via sd card and usb drive work as described above.

fvrmr, 2021-02-16:
Hi @ryanaxiom thank you for your feature request.

I have a work around that might work for you. 
You can use the post processing scripts: `pause at height` and `search and replace`. With the search and replace you could search for `M0` and replace it with `G4 S time in sec`

I have to defer this feature request because we don't maintain this firmware. Hope that this work around works for you.
Please let me know if you have more questions.

GregValiant, 2021-02-17:
I tried to chase this problem down because it occurred on numerous Creality machines with TFT displays and 4.2.X boards.  The gcode commands that send text to the display are ignored by certain printers.  In particular, M0, M1 and M117 are ignored.  If you were to try adjusting the "M0" to "M0 S180" then you might get 3 minutes to make the switch (if it works).  I can't test it because M0 works on my printer.  Other than trying that (or @fvrmr's search and replace fix) it looks like a firmware update is required.

GregValiant, 2021-03-18:
If this is still an issue I came across an alternative fix.  Instead of using the Marlin (M0) use BQ(M25).  The machine will complete the parking cycle and actually pause.  M25 usually needs an M24 to resume but you can fool the printer into sending one to itself.  When you are ready to resume use the LCD and select "Pause" and then "Resume".

closed 2023-07-13 by github-actions[bot]