Ultimaker/Cura #9274: creality ender 3 wont connect to my computer via usb. My usb ports work and the cord is good. It says the ender 3 is ready but nothing pops up in the monitor section

cchilders20, 2021-02-14:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-15:
Hi @cchilders20 could you please fill in the form and add a log file and maybe a screenshot. Thank you.

GregValiant, 2021-02-17:
"...the cord is good."  Is it a good phone _charging_ cable or is it a good computer _data_ cable?

firefighter331, 2021-02-20:
> **Actual results**
> (What happens after the above steps have been followed.)
> 
> **Expected results**
> (What should happen after the above steps have been followed.)

Shouldn't the "Expected results" come before the "_Actual results_?" It makes more sense to me that way. What should've happened, then what did happen.

Most people wouldn't say "_My phone did not charge_ when it was supposed to charge." Instead, they'd say "My phone was supposed to charge _but didn't_."

Just a suggestion that would take about 2 minutes to fix. I've been paying **wayyyy** too much attention in English class lol this needs to stop

closed 2021-03-01 by no-response[bot]