Ultimaker/Cura #9275: PinePhone UI scaling, and Touch Functionality to rotate work not working.

djhnsn, 2021-02-15:
**Is your feature request related to a problem? Please describe.**

I am using the mobian pinephone linux distribution which is based on debian.
Yes, this program works wonderfully on the pinephone except the ui doesn't scale well on mobile, and the touch functionality is limited to buttons and menu items. I can not rotate the mesh without a mouse. 

**Describe the solution you'd like**
It is my understanding that there are libraries from the mobile linux teams to correct these issues. I would like to see Cura be functional on a pocket computer. It otherwise works as expected, and works well when an external display and mouse are connected to the phone. 

**Affected users and/or printers**
All linux mobile users 

**Additional context**
It's pretty cool that I can create or download stl files on my phone. It's also pretty cool that I can slice them. It would be better if in a future update you can add the scaling and touchscreen fixes I mentioned. 

Thank you!

fvrmr, 2021-02-15:
Hi @djhnsn thank you for your feature request.

Really cool to hear that you can use Cura on your PinePhone. 

The UI depends on the GUI framework and Cura depends on the Qt framework. Currently we are on Qt 5.10. So we can't change our framework for mobile users. 
In the future we will update the Qt version and then it might work better. 

Found this reddit post about scaling on Qt applications: https://www.reddit.com/r/pinephone/comments/jvvn71/scaling_on_qt_applications/

There is already a post about the touch support, see here: #2515. 

But I will discuss the possibilities with the team. Keep you posted!

djhnsn, 2021-02-15:
Thank you for your reply 👍

fvrmr, 2021-02-16:
Hi @djhnsn I have discussed it with the team. 
But I have to defer this, because we can't build this just for one device now, so it is not one of our priorities. 

A pull request is to make this work is always welcome.

djhnsn, 2021-02-17:
> Hi @djhnsn I have discussed it with the team.
> But I have to defer this, because we can't build this just for one device now, so it is not one of our priorities.
> 
> A pull request is to make this work is always welcome.

That's reasonable. 
I don't actually know how to make this work, but it might be something worth learning for a few reasons. I am sure there will be more foss devices in the future, hopefully I figure it out. I will definitely look into it. 

Thanks for bringing it up.

closed 2023-07-13 by github-actions[bot]