Ultimaker/Cura #9276: Cura mysteriously drastically changing the shape of the model

CaptainSweatpants, 2021-02-15:
**Application version**
Cura 4.8.0

**Platform**
Win10

**Printer**
Custom Printer & Also same issue on a different Pursa MK3

**Reproduction steps**
1. (Exported Gcode from cura with perfect looking circles, but printing result added a bend & space in the circles. This bend & space is bigger and/or smaller depending on the space in between my walls; also is created with no space between walls. This distortion does not appear on Cura Gcode preview or Octoprint CGcode previews; but if you watch the video, you can see the printer is following that path; its not a printer slip or malfunction -- also tried on 2 printers, with the same result.

**Screenshot(s)**
[image]

** Pay no attention to my z-offset. I know it's too close. It was a result of my adjusting it when this started happening. I thought it may be a "too far", "Loose adhesion" problem at first. But after i saw the nozzle following the path of this bend, i knew it was doing it on purpose.

**Actual results**
The shape of the Circle has a "space/bend" added which changes the shape of the original circle.
Printing from any source (SD, Octoprint, Cura) makes no difference.
I printed this 7 times; each time adjusting line widths, fill gap settings, top/bottom patterns.. nothing changed anything regarding this. I printed this same model this morning flawlessly -- minus a 0.15mm tolerance added to the inner circle. after that,this started happening. 

**Expected results**
A Circle

**Project file**

All Files, Including Cura Project, Screenshots, and Video of issue: https://we.tl/t-yhrEHhVzA2 

**Additional information**
(Extra information relevant to the issue.)

CaptainSweatpants, 2021-02-15:
STL of Model can be found here if relevant
https://we.tl/t-d50T8BzWi4

fvrmr, 2021-02-15:
Hi @CaptainSweatpants thank you for your report.
I looked into your video and project file and it looks like it has something to do with your printer and not with Cura. 

Because if I look at your file you see that first the printer prints it clock wise but then it changes and goes counter clockwise. And if your pulleys are not tighten enough on the point where it changes it makes a straight line and then starts turning
You can find more about hysteresis in this reddit post: https://www.reddit.com/r/3Dprinting/comments/5ca6bq/hysteresis_circles/

So you could try tighten your pulleys, so it doesn't move while changing direction.

Please let me know if this works for you.

CaptainSweatpants, 2021-02-17:
Interesting read on Reddit; but a couple things don’t apply here. First thing is this is happening on 2 different printers, suddenly at the same time. The chances two belts could get loose at the same time is not likely. 

But I did confirm both belts are properly tensioned; enough for a little slack, but never loose enough to pinch and come close or touch each other. I also watched the drive pulley on both machines and the belts hug it nicely and no warp or slack on direction change. 

That Reddit post also refers to spring tensioner, which I don’t use. I use M5 idler pulleys screwed down. 

I took a shot in the dark and tried to reslice with another slicer app, and after 2hrs the circles kept their shape. Although I didn’t match up every setting in the other slicer, but I matched speed, line widths, flows. 

I’m having other issues with Cura and they seem to involve the “fill gaps between walls” feature; this may be another side effect to that possible main issue. Here’s a video of that issue, randomly traveling across a print to one spot and then going back, line after line. https://youtu.be/AVeUKlhk8FE 

I’m still putting the details into what I’ve found about that issue, then I’ll open a issue here about it.

GregValiant, 2021-02-17:
When I printed the model it was concentric to .01mm.  There were no oddities.  That was with Cura 4.8.
This is at a very basic level of Cura.  I'm sure if there was a problem making things round, thousands of people would be howling about it.  I understand that the issue occurs on two different printers but it does look mechanical, and I can't reproduce it.  The working diameter I assume was to be 14.00.  I get 14.10 in one direction and 14.11 in the other.  Understand that I'm a user and not affiliated with Ultimaker or Cura in any way.  If you would post a problem gcode file then someone could look at it or put it in a third party gcode analyzer (to take Cura out of the loop) and maybe tell if there is a problem in the generated gcode.  But as I've said, it sure has the look of a mechanical problem.

Ghostkeeper, 2021-02-24:
Maybe you changed acceleration settings when it started happening?

If Cura instructs the printer to make round movements, but the printer doesn't make round movements, I tend to put blame on the printer.

closed 2021-03-05 by no-response[bot]