Ultimaker/Cura #9278: [4.8.0] Pause at Height fails after model is rotated

RocketCityPinball, 2021-02-15:
**Application version**
4.8.0

**Platform**
Windows 7

**Printer**
Creality Ender-5

**Reproduction steps**
1. Imported model to build plate, added Post-Processing Code for Pause at Height (at layer), sliced, saved to disk.
2. Rotated existing model by 45-degrees, did not edit nor remove Post-Processing Code in place, sliced, saved to disk.

**Actual results**
1. Non-rotated model printed properly, pausing at proper layer height.
2. Rotated model did not pause at height and continued printing as if there were no post-processing code.

This is the first time I've run into this issue; I've been using Pause at Height frequently lately -- including with rotation -- and have never had a problem, even when using my various saved profiles.  During troubleshooting I've reinstalled Cura multiple times and have run nearly 20 different tests to try and isolate the issue.  It appears just rotating it is suddenly causing it to ignore the post processing code and I don't understand why.

RocketCityPinball, 2021-02-15:
Update: It doesn't appear to be based on rotation.  I imported the same model pre-rotated to 45 degrees and it still would not Pause at Height.  The only remaining variable is Supports.  I will try another test run with the rotated component but with no supports and see what it does.

Of course, if that turns out to be the case, it's still a serious problem because I need supports to print many items...

fvrmr, 2021-02-16:
Hi @RocketCityPinball thank your for your report.
Could you share your project files, gcode and log file? So I can try to reproduce your issue. 

Your log file can be found here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log `

RocketCityPinball, 2021-02-17:
Thanks for the followup.  Unfortunately the files and gcode contain proprietary data, so I'll have to generate other files which behave the same.  (I've been running a number of tests over the last day or so and confirmed the failure does happen with other files too.)

I performed further troubleshooting and dug more into the gcode, and found the trigger appears to be having Supports in the model AND printing on a Raft.  With both of these features active, the Pause at Height (M0) disappears from the gcode.  Turn either of them off (e.g., no supports, or using a brim instead), and the M0 code reappears.

fvrmr, 2021-02-18:
Hi @RocketCityPinball yeah other files that has this problem to are also very useful. Because I tried these steps and the M0 is still showing in my gcode. So a project file and gcode would help me trying to reproduce this.

closed 2021-03-04 by no-response[bot]