Ultimaker/Cura #9279: Combing travel paths seemingly not working

3781637, 2021-02-15:
**Application version**
(The version of the application this issue occurs with.)
4.8.0 & 4.7 (the 2 I've tried)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 with a GTX1080

**Printer**
(Which printer was selected in Cura?)
Tenlog Hands 2, and Creality CR10

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
N/A

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)
The travel paths are seemingly not following combing rules. 

**Expected results**
(What should happen after the above steps have been followed.)
I would think combing rules would have the travel paths at the lower layers following within the support areas and at the higher layers within the already printed areas. As it is now this doesn't seem to be working at all.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)
[gripFuzz.zip](https://github.com/Ultimaker/Cura/files/5983659/gripFuzz.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-16:
Hi @3781637 thank you for your report. 
You are suggesting that the travel paths are different between lower layers and higher layers. But that is not how combing works, it looks at the whole model.
Try to enable `avoid printed parts when traveling` and changing the value of `max comb distance with no retract`.

Please let me know if this makes any sense for you.

3781637, 2021-02-16:
No, I'm not suggesting that. Making adjustments to 'avoid printed parts when traveling' and 'max comb distance with no retract,' as well as related combing settings (all, not on skin, within infill), does not seem to change the behavior. The travel path does not stay within the printed area, and instead moves directly through/over the empty space in the center of the model. Throughout the majority of the top half layers it seems it should be possible for combing to work by moving around over the top of the already printed areas, but instead it cuts across the open space as if it doesn't recognize it as being open space. Earlier when I mentioned the lower layers I just meant that it would also be nice, in this case at least, if combing would recognize supports as printed areas so it would use the supports printed in the bottom half of this model as a travel path instead of traveling through the open space.

GregValiant, 2021-02-17:
I can see what you mean.  It's the way the model is assembled because fuzziness is only wanted on the outside of the grips.
Your model consists of 4 parts (5 because there are two grills).  The part "GripSmooth" is not watertight but that isn't likely a problem with combing.
It looks like within each part combing is correct.  And when traveling between parts, Cura correctly cuts across spaces to avoid the other parts.  That might cause a stringing problem.
I don't think you have a lot of options.  If you export the parts as a binary STL file all the parts will merge into a single part.  Then when you bring that new model back into Cura, you will see that the combing is as you expected, but you will have lost the ability to have the Fuzzy skin only on the outside of the grips.  Exporting everything but the "GripFuzz" part so the others get fused together helps the combing situation but Cura isn't an assembly modeler and it might be a pain to relocate that model within the GripFuzz part.  There would still be two parts and the combing still wouldn't be optimal - but it might be better than having 5 parts.

Ghostkeeper, 2021-02-24:
As far as I can tell, this issue arises from two requirements we have on the print order:

* All the support is printed before model.
* All parts of one model are printed before it starts with any parts from other models. I.e. parts are grouped by the model they belong to.

Since all the support is printed before the model, it needs to travel from one piece of support to another, which has some long travel moves across the front in your case. Since it's printing all the parts of an object clustered together, it's also travelling back and forth.

Cura will try to avoid going through the outer walls as much as possible if combing and Avoid Printed Parts When Traveling are enabled. But this only goes for parts that are already printed on that layer, not on parts printed on previous layers. So when it moves from the left grip to the right, it simply sees that it needs to pass outside of the left grip, travel a bit through air, then go into the right grip.

This is something that could theoretically be improved. We know the shape of the outer wall on the previous layer because even though it is multi-threading the layers, the walls are generated first in order to determine the areas of infill and skin.

3781637, 2021-04-16:
[image]
Still trying to figure out how to get the travel paths to work a bit better. I'm experimenting but even just using this one part I still cannot get Combing settings to work as expected. This is the same model just with the additional parts removed.

GregValiant, 2021-04-17:
Under "Travel" there is "Layer Start X" and "Layer Start Y" and in Shell you can set the Z seam location.  Tweaking those settings can effect the travel moves as well.  Having the layer start point near the Z seam (layer end point) might get rid of more of the combing moves.

3781637, 2021-04-19:
[image]
I've tried Cura 4.9.0 Beta and Arachne Engine Beta and there is seemingly no change in behavior

Ghostkeeper, 2021-05-04:
> Under "Travel" there is "Layer Start X" and "Layer Start Y" and in Shell you can set the Z seam location.

This is a mix-up. If the Z Seam Alignment is set to User Specified, the "Z Seam X" and "Z Seam Y" settings allow you to set the seam location. The Layer Start X and Layer Start Y settings determine which part of the layer it's going to start with. It will start with the part that is closest to those coordinates. Since the big fuzzy part envelops the smaller normal part almost completely, it'll almost always start with the big part. But it can still influence it.

So part of the problem is that Cura will try to print parts belonging to the same mesh together. One workaround could then be to add cutting meshes to the scene such that the left part of the fuzzy part becomes separate from the right part. It'll then simply print in the order that requires the smallest travel distance. However these cutting meshes cut the model, so you'll have a border in between then. It's all not ideal, but may be better for the moment.

3781637, 2021-10-30:
No change in behavior using Cura 4.11.0