Ultimaker/Cura #9280: Ausgleich Filamentfluss

lwbsbde, 2021-02-16:
CURA 4.8.0
Die "Maximale Geschwindigkeit für Flussausgleich" wird beim Slicen nicht übernommen.
The maximum speed for flow compensation is not adopted when slicing.

fvrmr, 2021-02-16:
Hi @lwbsbde could you please fill in the form and add screenshots, a project file and a log file. So we can help you properly.

lwbsbde, 2021-02-16:
[image]

fvrmr, 2021-02-19:
Hi @lwbsbde thank you for the screenshot. 
I will discuss it with the team if the maximum speed for flow doesn't affect the infill speed. 
Keep you posted.

fvrmr, 2021-02-22:
Hi @lwbsbde I have to defer this issue. Because it is correct that the `maximum speed for flow equalization` doesn't affect the `infill speed`. 
Hope you understand. Please reach out if you have more questions.

Ghostkeeper, 2021-02-24:
The Flow equalisation only affects the gap filling at the moment.

closed 2023-06-13 by github-actions[bot]