Ultimaker/Cura #9282: [4.8.0] Error: 'application/x-ufp' - Cannot export files as UFP or print over network

Teeffelen, 2021-02-17:
**Application version**
4.8.0

**Platform**
Operating System: Arch Linux
KDE Plasma Version: 5.20.5
KDE Frameworks Version: 5.79.0
Qt Version: 5.15.2
Kernel Version: 5.10.16-arch1-1

**Printer**
Ultimaker S5 - connected over network, updated to 6.1.1

**Reproduction steps**
1. Start Cura with `env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura` (see #5490)
2. Load up a file
3. Hit 'Slice'
4. Try to 'save as file' or 'print over network'

**Screenshot(s)**
[image]

**Actual results**
Displays error: 'application/x-ufp', and no file is exported/printed

**Expected results**
File saved as UFP file or printed over network succesfully

**Project file**
Cura crashes when trying to save a Project File, another bug?

**Log file**
Only exception raised when trying to 'print over network':
```
2021-02-17 14:54:29,470 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]: Exception: Unable to write to file UMS5_xyzCalibration_cube: 'application/x-ufp'
2021-02-17 14:54:29,471 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]: Traceback (most recent call last):
2021-02-17 14:54:29,472 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/python3.9/site-packages/UM/Qt/Bindings/OutputDeviceManagerProxy.py", line 150, in _writeToDevice
2021-02-17 14:54:29,473 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     device.requestWrite(nodes, file_name, limit_mimetypes, file_handler, **kwargs)
2021-02-17 14:54:29,474 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/LocalClusterOutputDevice.py", line 139, in requestWrite
2021-02-17 14:54:29,474 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     job = ExportFileJob(file_handler=file_handler, nodes=nodes, firmware_version=self.firmwareVersion)
2021-02-17 14:54:29,475 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/ExportFileJob.py", line 17, in __init__
2021-02-17 14:54:29,476 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     self._mesh_format_handler = MeshFormatHandler(file_handler, firmware_version)
2021-02-17 14:54:29,477 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/MeshFormatHandler.py", line 25, in __init__
2021-02-17 14:54:29,478 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     self._preferred_format = self._getPreferredFormat(firmware_version)
2021-02-17 14:54:29,478 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/MeshFormatHandler.py", line 113, in _getPreferredFormat
2021-02-17 14:54:29,479 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     file_formats = [format_by_mimetype[mimetype] for mimetype in machine_file_formats]
2021-02-17 14:54:29,480 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:   File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/MeshFormatHandler.py", line 113, in <listcomp>
2021-02-17 14:54:29,480 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]:     file_formats = [format_by_mimetype[mimetype] for mimetype in machine_file_formats]
2021-02-17 14:54:29,481 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [161]: KeyError: 'application/x-ufp'
```

**Additional information**
Installed Cura from [official Arch Repo's](https://archlinux.org/packages/community/any/cura/), the AppImage from official site does seem to work (and also fixes issues like #5490 on KDE).

fieldOfView, 2021-02-17:
> Log file
> Only exception raised when trying to 'print over network':

Yet the rest of the log will show that the UFPWriter module fails to load because of a missing dependency (libCharon). There are likely still other missing dependencies too. Please take this up with the maintainer of the official Arch repo package. It is not maintained by Ultimaker. Ultimaker only maintains and supports the AppImage for Linux.

Teeffelen, 2021-02-17:
I see, thanks for the help!

I will close the issue now as it is not a Cura bug 😄

closed 2021-02-17 by Teeffelen