Ultimaker/Cura #9283: when i am scrolling the list of options it appears and disappears.

qholmes229, 2021-02-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-18:
Hi @qholmes229 thank you for your report.
I am not able to reproduce this. Could you add your log file? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

qholmes229, 2021-02-18:
Here is the log file.

Curiously i was just going to make some pictures of it with the information
gone. Only when i scroll down. not up. But now if i wait a sec it actually
comes back. So not as much of a deal. just a bit annoying.

Thanks

Q

fvrmr, 2021-02-19:
Hi @qholmes229 I can't see your log file, because you replied on a email. Please add your log file in an comment on Github.

qholmes229, 2021-02-19:
Sorry here you go.

Q
[cura.log](https://github.com/Ultimaker/Cura/files/6010586/cura.log)

qholmes229, 2021-02-19:
Hey.. here is a picture of the missing info. if i dont drag the mouse over the details window it stays gone.

[image]

Q

Ghostkeeper, 2021-02-24:
Duplicate of https://github.com/Ultimaker/Cura/issues/8236 .

I had made a recording of the issue: http://dulek.net/work/setting_list_glitchy.webm

closed 2021-02-24 by Ghostkeeper