Ultimaker/Cura #9284: wont import obj file that used to work and works in other slicer

qholmes229, 2021-02-17:
**Application version**
4.8.0

**Platform**
win 7

**Printer**
Ender 5

**Reproduction steps**
1. try to import obj file i have used in the past. Just gives error. corrupt or missing file.
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-18:
Hi @qholmes229 thank you again for your report.
Could you share your obj file? And maybe add screenshots of the errors you are getting.

qholmes229, 2021-02-18:
Hmm i tried to add the file but it was too big? i was using email. i will try here.

Attached.. i think.. the error message i get and it opens in the Creality slicer and the file.

Thanks

Q

[image]
[image]

[hand bracket-02.zip](https://github.com/Ultimaker/Cura/files/6005405/hand.bracket-02.zip)

fvrmr, 2021-02-19:
Hi @qholmes229 thank you for sharing. I can't import your model either. 
Does it happen to other obj files to? Because I tried it myself with another obj file and that works fine. 

And could your share your log files? You can find your log file here: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

fieldOfView, 2021-02-19:
It seems the loading of this model broke in this commit: https://github.com/Ultimaker/Uranium/commit/34d04ddf0b600b17aa299382258deae9854e1227

It also seems the normals of this model are quite broken. But Cura should still handle the model.

qholmes229, 2021-02-19:
I am sure its an ugly model.. i did not make it. and just hacked it to make some fun brackets for my son. But it originally worked. And this time i used Creality slicer to save as STL and opened in Creality slicer.

Thanks 

Q

rburema, 2021-02-25:
Thanks for finding that @fieldOfView !
Hm, that commit (message) says:

> However it's not possible to have a face with vertices and normals but no textures:
>  f v1/n1 v2/n2 v3/n3

But that _should_ be possible by leaving the middle blank, like, in fact, in the users' file:
`f 1//1 2//2 3//3 `
(note the double forward slashes here!)

Ghostkeeper, 2021-02-26:
Normals are a bit weird in that model, but thanks to Ultimaker/Uranium#672 it now loads fine from the OBJ.
[image]

closed 2021-02-26 by Ghostkeeper