Ultimaker/Cura #9285: Artillery X1

gusdudu, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

Hellow, i need a profile for Artillery X1, it is not in the Listo of printers and it´s a most recognized printer for a lot of people like me! PLEASE include it thanks!!

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

**Additional context**
(Add any other context or screenshots about the feature request here.)

StuSerious, 2021-02-17:
@gusdudu the Artillery SideWinder x1 & its little brother, the Artillery Genius have been added in Cura 4.8 with this successful PR: [#8301](https://github.com/Ultimaker/Cura/pull/8301#issue-477688531).
Maybe next time try and provide more info, take the time to look at release notes or to actually fill the report out!
I think it's fair to say *everyone* would save time, you included!
Have a nice day!

fvrmr, 2021-02-18:
Hi @gusdudu please check out the PR that @StuSerious is suggesting.

Also good to know the Cura team doesn't maintain profiles from other printers (only Ultimaker) 
But we are very thankful to have a great community who maintain and submit those profiles. 
I will close this issue for now, but if you have questions, feel free to ask them here.

closed 2021-02-18 by fvrmr