Ultimaker/Cura #9286: [4.8] Changes to nozzle size didn't saved

mykola-corgi, 2021-02-17:
**Application version**
4.8

**Platform**
Windows 10 x64

**Printer**
Custom FFF Printer

**Reproduction steps**
1. Modify nozzle size for the machine in Machine Settings from 0.4 to 0.6.
2. Close menu
3. Back to the nozzle size settings

**Screenshot(s)**
[image]

**Actual results**
Nozzle size is still 0.4 

**Expected results**
Nozzle size is 0.6

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/5997878/cura.log)

fieldOfView, 2021-02-17:
Try using a comma instead of a dot. This has been fixed in the next version of Cura.

mykola-corgi, 2021-02-18:
Thanks, comma works like a charm

closed 2021-02-18 by mykola-corgi