Ultimaker/Cura #9287: Mesh Modifier moves in 3mf file

GregValiant, 2021-02-17:
Models are added to Cura.  A bed size mesh modifier is placed on the Cura build plate.  The project file is saved.
When the project file is opened the mesh modifier moves -25 in the Y.  When the file is again saved and re-opened the mesh modifer moves an additional -25 in the Y.  This happens every time the file is saved and re-opened.

**Application version**
4.8

**Platform**
Win10 Home in compatibility mode

**Printer**
Creality Ender 3 Pro

**Reproduction steps**
1. Load a model
2. Multiply the model +1
3. Add a 1mm thick x full build plate mesh modifier
4. Save the project file.
5. Open the project file in Cura.

**Screenshot(s)**
Image 1 as set up by user and then saved as a 3mf project file
[image]
Image 2 when re-opened in Cura.  The mesh modifier has moved -25 in the Y.  It was saved again as a 3mf project file
[image]
The second save opened in Cura.  The mesh modifier moved an additional -25 in the Y.
[image]

**Expected results**
Mesh modifiers should stay where they are put.

**Project file**
[modifier.zip](https://github.com/Ultimaker/Cura/files/5999179/modifier.zip)

**Log file**
N/A

**Additional information**
Reported by Curasurf on the Ultimaker Forum.

fvrmr, 2021-02-18:
Hi @GregValiant thank you for your report.
I am able to reproduce this. But it doesn't happen when I try this on an Ultimaker 3. 

I will discuss this with the team. Keep you posted.

GregValiant, 2021-02-18:
I think it's that specific file.  When creating my own file with the same conditions the modifier doesn't move.  When changing to an S3, S5, 3, CR-10, Ender, etc., saving, and re-opening that specific file, the modifier moves.  @curasurf on the Ultimaker forum created that file.  Maybe he will respond here and shed some light.

fvrmr, 2021-02-19:
@GregValiant I tried it on a UM3 with the same file. And now @rburema and I can reproduce this also on a UM3. And it also happens when you resize the modifier. Thought that would cause the problem.

I looked into if it is file specific. And it looks like that that is the case. Because I couldn't reproduce this with my own files, as I see you tried to.
When I delete the modifier and add a new modifier it doesn't happen either. I will ask @curasurf to try that on the form. 

Thank you @GregValiant for you effort in this issue (and all the others issues off course)

Ghostkeeper, 2021-02-25:
I inspected that 3MF file but the document is really nothing special. There's two completely separate models. They each have a transformation, e.g. moving them to the middle of the build plate. The actual model seems to be mirrored in the X and Z direction. The mesh modifier is only rotated.

Creating a new scene, we weren't able to reproduce this yet, not even when trying to get similar transformations on the objects.

GregValiant, 2021-02-25:
Since it is file specific I'll take the liberty of closing the issue.  That is the only file I have seen with the problem as well.

closed 2021-02-25 by GregValiant