Ultimaker/Cura #9290: crash on save project (i18n)

deBugger-, 2021-02-18:
to reproduce
- create new project (i18n is english)
- add any change
- save project (works fine)
goto preferences -> Configure Cura -> General Language -> chose Deutsch > close 
restart cura

- Create new project i18n ( Note: language is german) ( Help to navigate: Datei -> Neues Project  )
- add any change
- save Project ( Help to navigate: Datei -> Project speichern ...)
-> failt in a crash
see @ ERROR 

restart Cura
goto preferences -> Configure Cura -> General Language -> chose English > close 
restart Cura
- Create new project (note i18n is back in english)
- add any change
- save project 
-> everthing is fine

ERROR:
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 1063, in event
    return super().event(event)
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 464, in event
    event._function_event.call()
  File "/usr/lib/python3.9/site-packages/UM/Event.py", line 218, in call
    self._function(*self._args, **self._kwargs)
  File "/usr/lib/python3.9/site-packages/cura/Utils/Threading.py", line 34, in _handle_call
    ico.result = func(*args, **kwargs)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 57, in write
    self._writeContainerToArchive(container, archive)
  File "/usr/lib/cura/plugins/3MFWriter/ThreeMFWorkspaceWriter.py", line 154, in _writeContainerToArchive
    serialized_data = container.serialize(ignored_metadata_keys = ignore_keys)
  File "/usr/lib/cura/plugins/XmlMaterialProfile/XmlMaterialProfile.py", line 154, in serialize
    builder.start("metadata") # type: ignore
TypeError: start expected 2 arguments, got 1

___________
Thread 0x00007fa0e5ffb640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 2530 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1735 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fa0e67fc640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/threading.py", line 574 in wait
  File "/usr/lib/cura/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fa0e77fe640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 316 in wait
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1485 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fa0e7fff640 (most recent call first):
  File "/usr/lib/python3.9/site-packages/zeroconf/__init__.py", line 1359 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fa0ecd39640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
[... 260 lines cut ...]

Thread 0x00007fa18f7ce640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0x00007fa18ffcf640 (most recent call first):
  File "/usr/lib/python3.9/threading.py", line 312 in wait
  File "/usr/lib/python3.9/threading.py", line 443 in acquire
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.9/site-packages/UM/JobQueue.py", line 123 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Current thread 0x00007fa1a8326740 (most recent call first):
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.9/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 167 in exceptHook
  File "/usr/lib/python3.9/site-packages/UM/Qt/QtApplication.py", line 378 in exec_
  File "/usr/lib/python3.9/site-packages/cura/CuraApplication.py", line 847 in run
  File "/usr/bin/cura", line 235 in <module>

fieldOfView, 2021-02-19:
The issue has been fixed for the next release of Cura: https://github.com/Ultimaker/Cura/pull/9105. Until then, use the AppImage. This is due to a change in Python 3.9. Cura 4.8 is not yet compatible with Python 3.9.

fvrmr, 2021-02-19:
Hi @deBugger- please try the AppImange @fieldOfView is suggesting. 
I will close this issue, because that will solve your problem.
But please reach out if you have questions or need more help.

closed 2021-02-19 by fvrmr