Ultimaker/Cura #9291: Cura Arachne. Program closes when ejecting the SD card.

peterp194, 2021-02-18:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-19:
Hi @peterp194 could you please fill in the form. Describe your reproduction steps, OS and add a log file. 
You can find your log file here on Windows: `C:\Users\\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
OSx: `$USER/Library/Application Support/cura/<Cura version>/cura.log`
Linux/Ubuntu: `$USER/.local/share/cura/<Cura version>/cura.log`

fieldOfView, 2021-02-20:
Duplicate of #9057

closed 2021-02-25 by Ghostkeeper

konskarm, 2021-04-08:
We just released the 4.9 beta. @peterp194 can you give it a try to see if the issue persists? So far we haven't been able to reproduce it in the 4.9 beta release.

peterp194, 2021-04-09:
Hi
The card ejection problem is with Cura Aracne Alpha and Beta not 4.8 or
4.9, they are ok.
Regards
Peter

Ghostkeeper, 2021-04-16:
Indeed, because the issue didn't exist yet in 4.8, and was fixed in 4.9 after Arachne was last split off.