Ultimaker/Cura #9292: Support brim limited to limited brim width

tlhintoq, 2021-02-19:
**Application version**
Arachnid beta

**Platform**
Windows 64

**Printer**
Hictop D3Hero

**Reproduction steps**
Turn no supports
Turn on brim
Turn on support brim
Set regular brim to 0 or 1 mm
Set support brim to 10mm
{because model is fine, but supports need help adhering}

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
Support brim maxes out at regular brim value

**Expected results**
Regular brim of 0mm
Support brim of 10mm

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save Project. For big files you may need to use WeTransfer or similar file sharing sites. G-code files are not project files!)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Asterchades, 2021-02-19:
I believe the Support Brim is for a brim that goes _under_ the support structure, not out to the sides. I would agree with you, though, that it doesn't really make a heap of sense to only do that without a more specifically worded distinction.

tlhintoq, 2021-02-19:
> I believe the Support Brim is for a brim that goes _under_ the support structure, not out to the sides. I would agree with you, though, that it doesn't really make a heap of sense to only do that without a more specifically worded distinction.

That doesn't make sense or follow any of the other naming conventions.
"Under" is a 'raft'
"Brim" starts at the edge of an object and goes outward.  Basically a 'brim' is a 'skirt' with an offset of zero.

fvrmr, 2021-02-19:
Hi @tlhintoq thank you for your report.
What @Asterchades is suggesting is correct. This is expected behaviour of Cura.
See the explanation of the `enable support brim` settings:
[image]

Hope this makes sense to you.

tlhintoq, 2021-02-21:
You can call it "expected" - but that doesn't make it 'right'.
The very definition of brim is "around an object".  Is the 'brim' for the model (not support) considered 'under but not around'?
If even the support brim extends 10mm around the support, is that not actually "around" and not "under"?

A *raft* is under
A *brim" is around

And look at the use-case here.  The model is fine without a brim.  Its not needed.   But the area to be supported is narrow.  The support is like 2mm wide by 30mm long.  *that* needs a brim to keep it in place.

Having to band-aid by putting  a 10mm brim around all of it just makes for a bunch of extra cleanup work.

Beside... Call it whatever you like... Would it be such a bad idea to make it possible to set the support brim and the model brim to be different values?

fvrmr, 2021-02-23:
Hi @tlhintoq I totally understand you that this is a little weird

Let me explain how the support brim works. 
One brim line consists of a polygon. And when you are printing with a `support brim` the changes are that the support brim will interrupt your regular brim. Because both brims are a polygon, you can't have a brim that's interrupted. 
We are aware of this, but for now it is not a priority to fix. 

Hope you understand.

Ghostkeeper, 2021-02-25:
Duplicate of https://github.com/Ultimaker/Cura/issues/3824.

closed 2021-02-25 by Ghostkeeper