Ultimaker/Cura #9294: Z Seam Alignment doesn't work in modifiers

PoiMechanic, 2021-02-19:
**Application version**
4.8

**Platform**
Macbbok Pro 16". Catalina. AMD PRO 5600m

**Printer**
Prusa Mk3s
**Reproduction steps**
Single model with part of it having a modifier which has z seam set to Front.
Main menu option z seam set to random.

**Screenshot(s)**
[image]
[image]

**Actual results**
Main menu settings overrides the modifier.

**Expected results**
I would expect the main menu setting to apply to any section without a modifier but that doesn't work. The lower part of the model with modifier is set to have seam at front and the rest needs random but I only get random. Happens on any model with any profile. 
**Project file**

[Modifier Issues.3mf.zip](https://github.com/Ultimaker/Cura/files/6007119/Modifier.Issues.3mf.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/6007128/cura.log)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-19:
Hi @PoiMechanic thank you for your report. 
I will discuss this with the team. 
But I think that the problem is that the option Z Seam Alignment shouldn't be visible at all in modifier settings. 
Keep you posted.

fvrmr, 2021-02-22:
Hi @PoiMechanic I have discussed it with the team. 
But I have to defer this, because if you use a cutting mesh, it will use a add the z seam at the front and the rest of the model random.

Please let me know if that works for you.

PoiMechanic, 2021-02-22:
I have tried this many times but it never worked but now I'm realizing it was always on 4.7. I just verified that it doesn't work on 4.7 but I just tried this on 4.8 and it is working. Thank you so much and sorry to waste your time.

closed 2021-02-25 by Ghostkeeper