Ultimaker/Cura #9295: layer issue

lbe001, 2021-02-19:
**Application version**
4.8

**Platform**
Win10

**Printer**
Artillery X1

**Reproduction steps**
Always

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
Pb offset the layers

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

fvrmr, 2021-02-19:
Hi @lbe001 could you share your project file? So I try to reproduce this issue.

lbe001, 2021-02-21:
Hi here it is, with CURA 4.7.1 I don't have this issue
Thanks
[anneau2.zip](https://github.com/Ultimaker/Cura/files/6018439/anneau2.zip)

fvrmr, 2021-02-23:
Hi @lbe001 thank you for your file. You attached a stl and not a project file. 
But this is the result I get in 4.8. So no missing layer. 

[image]

Could you add your project file (file - save project) a project file contains all your settings. So probably also the missing layer.

lbe001, 2021-02-24:
Hi, here is the project file

[anneau2.zip](https://github.com/Ultimaker/Cura/files/6039129/anneau2.zip)

fvrmr, 2021-02-26:
Hi @lbe001 thank you for your project file.

This is happening because of the `hole horizontal expansion`. If you look with the top view at your model you can see that this is happening at layer 28 and layer 49 and between these layers there's is no horizontal hole, because of a 'gap'.  

Layer 48:
[image]

As you can see in the screenshot above is there is no 'hole' , because of the hole in the wall. And at the next layer there is a hole, so the hole horizontal expansion applies.

Layer 49:
[image]

When I change the hole horizontal expansion to 0 you will see that it won't happen.

Unfortunately the only solution for this is changing your model in a CAD program and add the expansion on your model there.

I will close this issue, because it is expected behaviour and not a bug. 
But please reach out if this makes sense to you or you have questions about this.

closed 2021-02-26 by fvrmr