Ultimaker/Cura #9300: slowly raise temperature based on mass

shimondoodkin, 2021-02-20:
a large print gets separated from the bed and wraps at the edges

although there is a pid for temperature it is in the middle. so edges are cold but center heat is maintained. it if will be a  little hotter it will still be ok.

from my best understanding the large part works better as a heat sink and dissipates temperature from the bed better
so bed becomes cooler at the base

the solution is raise temperature slowly

a sophisticated solution is to calculate based on material heat conductivity. maybe there is some code to calculate 3D object heat transfer.

heat at bed center - heat lost from heat sink = heat at edges. 

and put in gcode command to set bed heat at certain percentages like every 10%.. or every x mass add.  and set: wanted bed temperature = base temperature + heat lost from the heatsink.

simple solutions:

to raise the temperature based on percentage and a temperature curve may be based on mass or manual chart.

to raise the temperature based on percentage and a range.

the next step is to automate it based on material library.

everyone with an fdm printer will benefit from this

fvrmr, 2021-02-22:
Hi @shimondoodkin thank you for your feature request.
I really like the idea you have. I will discus this with the team. Keep you posted. 

And for next time, please fill in our form correctly, that helps us to understand feature request better.

shimondoodkin, 2021-02-22:
thank you for liking the idea.

a possible build up for the idea for the nozzle too, probably insignificant though:

a hypothesis: maybe the model also takes heat from the nozzle. but it is only a hypothesis, should be checked.

last time I printed a 50g petg box of 2mm shell, with 3x7x15hmm sticks at top and bottom.  I noticed stress in the print. the sticks at top were more brittle and broken more easily, but I can't blame for sure that the reason was stress in the print. 

a guess is maybe it is because of dragging a cold melted filament from the nozzle and stretching it.

fvrmr, 2021-02-23:
Hi @shimondoodkin thanks for the reply.
But I have to reject this issue. I have discussed it with the team and we agreed that this is more depending on the printer settings.
And adding this feature would also be very difficult to understand for other users and probably not many users will use this.

A few work around you can use to avoid warping are: 
- Better adhesion, something like tape or glue.
- Printing with a brim or raft.
- Change `top/bottom pattern` to `concentric`. 

Please feel free to reach out if you have more questions.

closed 2021-02-23 by shimondoodkin