Ultimaker/Cura #9302: Material management. What happen on cura 4.8? its not working with several printers

ldiegos, 2021-02-20:
**Application version**
Cura 4.8

**Platform**
Windows 10

**Printer**
Creality ender 3 pro
Geeetech A10M

**Reproduction steps**
1. Create a material on ender 3 pro and give all the stuff, temperature, retractions, and so on.
2. Change to geeetech A10M, configure the material to print also on it.
3. Return to ender 3, all the configuration has gone and replaced with the A10M.

**Screenshot(s)**
Created and configured material on Ender 3 Pro.
[image]

Change printer to A10M. The material starts with the same configuration as the E3Pro.
[image]

Configure the material in A10M.
[image]

Return to E3Pro. BAAAANG configuration of the A10M. 
[image]

**Actual results**
See screenshots.

**Expected results**
Same as 4.7. The configuration of the materials remain independent with the printer.

**Project file**
N/A

**Log file**
N/A

**Additional information**
On version 4.7 when access to your cura profile information about the materials, the files where having the different printers configuration. One section was global for all non listed printers, but other printers had their own configuration. And when i was changing the configuration on the ender 3 Pro, the config section of the ender 3 pro, was changing. 

On the cura 4.8, it ignore all the configuration sections and start changing the global one, and this is a completely mesh and lost of time and filament and effort.
```
  <settings>
    <setting key="standby temperature">200</setting>
    <setting key="retraction amount">6.5</setting>
    <setting key="print temperature">200</setting>
    <setting key="retraction speed">35.0</setting>
    <setting key="adhesion tendency">0</setting>
    <cura:setting key="material_flow">85.63</cura:setting>
    <setting key="build volume temperature">28</setting>
    <setting key="surface energy">100</setting>
    <setting key="heated bed temperature">35</setting>
    <machine>
      **<machine_identifier manufacturer="Creality3D" product="creality_ender3pro" />**
      <setting key="retraction amount">6</setting>
      <setting key="retraction speed">40</setting>
      <cura:setting key="material_flow">98.8</cura:setting>
      <setting key="standby temperature">210</setting>
      <setting key="heated bed temperature">50</setting>
    </machine>
</settings>

```

When trying to create a new filament, the configuration for the "creality_ender3pro" is missing, and if i try to add it manually is completely ignore.

Why is this happening?
Why don't you just add all the printers when creating a new filament to the configuration file? and not only a few of them.

Can you add the model of the printer and also the name of the given printer? Because is it possible to have the same printer several times and each printer have slightly differences in the parameters.

This could be done on the fly, when creating a filament, add the model, the name and all the settings for it.
When changing the printer, detects the model and name and change the settings or create the setting for this new filament.

example:
```
<?xml version='1.0' encoding='utf-8'?>
<fdmmaterial version="1.3" xmlns="http://www.ultimaker.com/material" xmlns:cura="http://www.ultimaker.com/cura">
  <metadata>
    <name>
      <brand>Generic</brand>
      <material>PLA</material>
      <color>Generic</color>
      <label>PLA</label>
    </name>
    <description>Fast, safe and reliable printing. PLA is ideal for the fast and reliable printing of parts and prototypes with a great surface quality.</description>
    <color_code>#ffc924</color_code>
    <version>8</version>
    <GUID>e35e719a-98aa-410b-a6be-f94f7e61ea33</GUID>
    <adhesion_info>Print on bare glass. Use tape for cold build plates.</adhesion_info>
  </metadata>
  <properties>
    <diameter>1.75</diameter>
    <density>1.24</density>
  </properties>
  <settings>
    <setting key="standby temperature">100</setting>
    <setting key="print temperature">100</setting>
    <setting key="surface energy">100</setting>
    <setting key="heated bed temperature">100</setting>
    <setting key="retraction speed">100</setting>
    <setting key="print cooling">100</setting>
    <setting key="build volume temperature">28</setting>
    <setting key="retraction amount">100</setting>
    <setting key="adhesion tendency">0</setting>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_x5sa_500" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10spro" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_xy2" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_x5sa_400" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10s5" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Lotmaxx" product="lotmaxx_sc60" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_d01" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_xy2pro" />
    </machine>
    <machine>
      <machine_identifier manufacturer="IMADE3D" product="IMADE3D JellyBOX" />
      </hotend>
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender3" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_xy3" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr20pro" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender5" />
    </machine>
    <machine>
      <machine_identifier manufacturer="BIQU" product="biqu_b1" />
    </machine>
    <machine>
      <machine_identifier manufacturer="CreawsomeMod" product="creality_base" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_x5sa" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender4" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender2" />      
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10s4" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Tronxy" product="tronxy_x" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Kati Hal ARGE" product="alyanx3dp" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr20" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10s" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Kati Hal ARGE" product="kupido" />
    </machine>
    <machine>
      <machine_identifier manufacturer="BIQU" product="biqu_b1_abl" />
       </machine>
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_cr10mini" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Kati Hal ARGE" product="alya3dp" />
    </machine>
    <machine>
      <machine_identifier manufacturer="Cartesio bv" product="cartesio" />
      </hotend>
    </machine>
  </settings>
</fdmmaterial>
```

Kind regards, 
Luis.

fvrmr, 2021-02-22:
Hi @ldiegos thank you for your report.
I can reproduce this. I will discus this with the team. 
Keep you posted!

ldiegos, 2021-02-22:
Thank you so much!!! 
i returned to to 4.7.1 while this is in study by the team. 
I like very much the filament management and this is one of the things that Cura surpasses other slicers and i don't want to  miss it.

Keep doing Cura the best slicer program!
Luis.

fvrmr, 2021-02-23:
Hi @ldiegos I tried it also on 4.7.1 and it also happens there. Can you show me how you do it in 4.7.1.
It is probably related to the printer and not something we don't maintain. Are you sure that is related to the material settings and not the profile settings. Because in the profile settings you can set settings per printer and this won't change because it is related to the printer.

Hope this makes sense to you.

ldiegos, 2021-02-23:
Hi,
I migrate the materials profiles since version 4.3/4.5, i don't remember, but this has been working since then, maybe at that moment i change something or i import a material from other site, but, this was working successfully.

Lets see, i have all of these materials in cura 4.7.1:
[image]

All of them has the global-non listed printer configuration and a section for "ender3_pro" configuration.
Global:
[image]

Ender 3 pro:
[image]

As you can see, and something that i imagine is that the CE3Pro configuration only stored the differences with the global section.

So if i open Cura 4.7.1 and access to the material with the CE3Pro, you can see the same information that in the material config file:
[image]

And when i change the printer:

ldiegos, 2021-02-23:
Despite that is is an issue or not... 

I'm wondering if you could take a minute to thing about what i proposed.

Add to the materials configuration files, the current configuration section for the users printers, so when i create a new material(cloned) or brand new the file <material>.xml.fdm_material automatically will be filled with your printer model and name.

Something like;
 <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender3pro" **name"ldiegoes_CE3Pro"** />
      <setting key="standby temperature">205</setting>
      <setting key="print temperature">205</setting>
      <cura:setting key="material_flow">98.8</cura:setting>
    </machine>
and 
    <machine>
      <machine_identifier manufacturer="Geeetech" product="geeetech_A10M" **name="A10M_multicolor"**/>
      <setting key="standby temperature">195</setting>
      <setting key="print temperature">195</setting>
      <cura:setting key="material_flow">75</cura:setting>
    </machine>

When you have two printers named: 
ldiegoes_CE3Pro
A10M_multicolor

And if you add a new one to your farm, named ABS_CE3Pro, then:
    <machine>
      <machine_identifier manufacturer="Creality3D" product="creality_ender3pro" **name="ABS_CE3Pro"**/>
      <setting key="standby temperature">280</setting>
      <setting key="print temperature">280</setting>
      <cura:setting key="material_flow">50</cura:setting>
    </machine>

Ghostkeeper, 2021-03-01:
Yeah, this is expected behaviour. Indeed I'm getting the same as Fvrmr: It worked the same as in version 4.7.1. In 4.7 I'm also seeing the material update when viewed from Ender 3 after updating it while the Geeetech printer was active.

The expected behaviour is that if there is a specialised sub-category in the material profile, editing the print settings in the GUI will edit the sub-category. If there is not, then editing the print settings will edit the "base" material profile. Neither the Ender 3 Pro nor Geeetech A10M have specialised sub-material profiles in Generic PLA 1.75mm, so editing the material in the GUI will edit the base material profile, and your changes will also show up for other printers that use the base material profile.

The logic behind this is that if a printer doesn't have any particular oddities, the base material profile is assumed to be the most accurate. If the material profile needs to be edited for one printer without particular oddities, then it likely needs to be edited for others as well. If the printer **does** have particular oddities, it will already have a sub-category in the material profile.
The property of whether a printer has particular oddities or not cannot be edited in the GUI, because that would be very complex to understand for the user. Changing that is in the domain of the profile designers, so you have to edit the XML files for that then.

A lot of people also depend on this behaviour, expecting that a material profile is just a single set of settings and changing it while one printer is active will always change it for the other printers too. So your proposal is not something we'd like to implement.

> All of them has the global-non listed printer configuration and a section for "ender3_pro" configuration.

That is strange, because the material you showed in your reproduction steps doesn't have a section for the Ender 3 Pro: https://github.com/Ultimaker/fdm_materials/blob/master/generic_pla_175.xml.fdm_material

Did you edit that in manually? NONE of the built-in material profiles in Cura refer to the Ender 3 Pro as of this writing.

> To add a new material, i'm realizing that always clone a existing one and remove the "linked" button. Maybe this is also has something to be take into account and i never think that the material configuration goes wrong in cura 4.7.1 also.

The "linking" of materials concerns only some of the metadata (spool cost, etc) and how the material gets matched with spools on the printer for some printers. The settings are not shared.

closed 2021-03-01 by Ghostkeeper

ldiegos, 2021-03-01:
Temperatures on A10m and ender3 are not the same for the same material, neither the retractions, so... i don't want to remember each time that i change the printer which temperature, retractions and all the stuff, that has each material.

And for sure not to create a new material profile for each printer, when i see in the xml file that you have a section for each printer, despite your explanation, that i don't understand very well ;)

so... i'm stuck on version 4.7.1.. because version 4.8, not work as the 4.7.1.

Thank you so much, and so sorry to hear this...
It't time to give a try on other slicers

Ghostkeeper, 2021-03-02:
I suggest you duplicate the material then for your two configurations of it.

ldiegos, 2021-03-02:
Yeah, that's what i wrote that is completely nonsense,

Thank you so much, see you
Luis.