Ultimaker/Cura #9304: Thin wall problems

atsju, 2021-02-21:
**Application version**
Cura 4.8.0

**Platform**
Win10-64

**Printer**
No preset. Manually defined CTC byzer frankenstein
0.4mm nozzle, Marlin 2.0, 1.75mm filament

**Reproduction steps**
1. Slice these 0.4mm single wall model (50mm diameter)
[circle.STL.txt](https://github.com/Ultimaker/Cura/files/6017361/circle.STL.txt)
2. The defect will appear at print

**Screenshot(s)**
[image]

**Actual results**
There are holes in wall during print.

**Expected results**
A smooth single wall print.

**Project file**
[CFFFP_circle.zip](https://github.com/Ultimaker/Cura/files/6017372/CFFFP_circle.zip)

**Log file**
Not relevant

**Additional information**
I'm unsure if this may be linked to #7656. I just mention it so you can double check.

To understand why there are 2 pass over the circle I made a square model.
[square.STL.txt](https://github.com/Ultimaker/Cura/files/6017360/square.STL.txt)
This is one typical layer of square gcode with relative extrusion.
My understanding is that instead of printing the line in 1 pass there are 2 pass and the second one has really low extrusion over the wall length (1.62um over 50mm is not "printable") plus one small blob in corner. This should all be optimized out to print in 1 pass I suppose.
```
;TIME_ELAPSED:73.151149
;LAYER:4
;TYPE:WALL-OUTER
;MESH:square.STL
G1 F1500 E6.5
G1 F2400 X74.8 Y90.2 E1.25723
G1 X125.2 Y90.2 E1.25723
G1 X125.2 Y39.8 E1.25723
G1 X74.8 Y39.8 E1.25723
G0 F7200 X74.8 Y40
G1 F1500 E-6.5
G0 F7200 X125.3 Y39.8
G1 F1500 E6.5
G1 F2400 X125.3 Y90.2 E0.31431
G1 X125.3 Y90.3 E0.00162
G1 X125.2 Y90.3 E0.00162
G1 X74.8 Y90.3 E0.31431
G1 X74.7 Y90.3 E0.00162
G1 X74.7 Y90.2 E0.00162
G1 X74.7 Y39.8 E0.31431
G1 X74.7 Y39.7 E0.00162
G1 X74.8 Y39.7 E0.00162
G1 X125.2 Y39.7 E0.31431
G1 X125.3 Y39.7 E0.00162
G1 X125.3 Y39.8 E0.00162
G0 F7200 X125.3 Y40
G1 F1500 E-6.5
;MESH:NONMESH
G0 F600 X125.3 Y40 Z1.05
G0 F7200 X74.8 Y39.8
```
The square shows really less hole than circle so the bug should focus on why the circle is making 2 pass. Square was mainly for my understanding.

fvrmr, 2021-02-22:
Hi @atsju thank you for your bug report. 
I will discus this with the team. Keep you posted.

fvrmr, 2021-02-23:
Hi @atsju I have discussed it with the team. 
And this is because the circle has an inner and an outer wall. That's why it passes 2 times.
With our Arachne release this is not an issue anymore, read [here](https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_alpha) more about Arachne. 

As you can see in our screenshot below this the circle is only printed in one pass.
[image]

atsju, 2021-02-23:
To be completely honest, I was about to definitively change Slicer software but I did some test with this and other pieces and the new engine is just working fine with thin walls now. You can close issue. Thank you.

fvrmr, 2021-02-26:
Hi @atsju nice to hear. I will close this issue!

closed 2021-02-26 by fvrmr