Ultimaker/Cura #9306: [4.8.0] Per model infill override affects other model

mulder999, 2021-02-21:
**Application version**
4.8.0

**Platform**
Win 10x64 v1909

**Printer**
AnyCubic i3 Mega S

**Reproduction steps**
1. Load a 3MF file with at least 2 parts
2. Ungroup model, you now have 2 parts: A and B
3. Choose 15% infill ratio with cubic subdivision (global setting)
4. Set 100% infill ratio for part B (per model setting)
5. Slice

**Screenshot(s)**
[image]

**Actual results**
* Part A uses about 90% infill / triangle subdivision (WRONG)
* Part B uses 100% infill

**Expected results**
* Part A uses 15% infill / cubic subdivision
* Part B uses 100% infill

smartavionics, 2021-02-21:
A workaround is to use a different infill pattern for part B, e.g. lines.

smartavionics, 2021-02-21:
As far as I can tell, this bug is specific to the cubic-subdivision infill pattern as you can use other patterns OK.

smartavionics, 2021-02-21:
Yes, I looked at the code and it appears that it can only support a single density of cubic-subdivision infill so all objects that use that infill pattern must use the same density.

mulder999, 2021-02-21:
Indeed, the workaround is working, for eg cubic infill works as intended. Only cubic-subdivision seems affected.

fvrmr, 2021-02-22:
Hi @mulder999 thank your for your bug report. 
Thank you @smartavionics for suggesting the work around and that this works for you. But I will discus this with the team.
Keep you posted.

fvrmr, 2021-02-23:
Hi @mulder999 I have discussed it with the team. Cubic subdivision isn't used by that many users, so it is not a priority for us right now. And because there is a work around I will defer this issue for now. 
Hope you understand.

closed 2023-06-13 by github-actions[bot]