Ultimaker/Cura #9308: Object intersects with the bed after rotating

sjoerdtimmer, 2021-02-21:
**Application version**
4.8.0

**Printer**
Ender 3 Pro

**Reproduction steps**
Open a large object like this one: https://www.thingiverse.com/thing:2446648
Move and rotate around until it fits the build area.
It is possible that after rotating the model intersects with the buildplate.
If this is only 1 or 2 mm it is very easy to miss that part of the model is simply not printed.
No warning or error is shown and neither did "drop to buildplate" kick in.

**Screenshot(s)**
[image]

**Actual results**
The object intersects the buildplate without any error or warning.

**Expected results**
It might (in rare cases) be desirable to cut off the first few layers this way, but in general this should at the very least cause a flashy warning message.

fvrmr, 2021-02-22:
Hi @sjoerdtimmer thank you for your bug report. 
Is in your preferences the setting `automatically drop models to the build plate` disabled? If it is, try enabling this setting.

But if you look closely at your model, you can see that a part of your model is blue. And that's the 'warning' you get. 
And yes in some cases you want this option. That's why you won't get a flashy warning message. 

Please let me know if this makes sense to you.

sjoerdtimmer, 2021-02-22:
Automatic drop-to-buildplate plate is enabled.
An "automatic **lift**-to-buildplate" option would indeed be another viable solution.

Before you classify this as "it's a feature, not a bug, I would like to argue that the use case for this feature is very rare (and a sign of bad design) and the annoyance of making a print with 5 missing layers from the bottom is much greater if it hapens. 

Let me also point out that I did not intentionally drag my model into the buildplate; It ended up that way after rotating. So, yet another solution is to do a lift-to-build-plate after each rotation(probably also scale, etc.). Someone could then still proceed to move the model down if they really wanted to.

Regarding the visual cue: The blue outline is not very noticeable. In my screenshot you can see it clearly because it's taken from the side but otherwise it is very easy to miss; in particular if only a few layers are below the bed.

sjoerdtimmer, 2021-02-22:
Further experimentation shows that the "auto-lift-to-buildplate" already happens after rotating but only if the model fits inside the build volume at that time. 

In my case the model was too large and imported to the side of the buildarea altogether. I rotated it (ending up just a few layers below the buildplate) and moved it onto the buildplate. 

So the true bug in this case is that the "lift-to-buildplate after rotate" does not happen if the model is not fully contained inside the build volume. I cannot think of any reason why this would make a difference. 

My suggestion for a fix would be to lift the model to the imaginary plane going through the buildplate after rotate/scale regardless of whether or not it fits the build volume.

fvrmr, 2021-02-23:
Hi @sjoerdtimmer thanks for your replies.
I understand what is happening. When I drag your file in to Cura and rotate the object 105 degrees and then I click Lay flat and it doesn't go into the build plate.

But I agree that sometime this is happening.
Although this is really not a bug, it is designed like this. I will discuss your other report with the team. 

And _automatic lift-to-buildplate_ could also be a feature request, so please fill in a new issue for that. 

I will close this issue, but feel free to reach out. Hope you understand that.

closed 2021-02-23 by fvrmr

sjoerdtimmer, 2021-02-23:
Before I open a feature request, can I just ask what the rationale is behind the difference in behaviour between rotations inside and outside build area?

fvrmr, 2021-02-23:
Yeah off course. The difference between rotating in the inside and outside build volume is that inside the build volume the object snaps to the build plate while rotating and outside this is not relevant, so it won't snap to a build plate because there is no build plate at all.

Ghostkeeper, 2021-03-01:
The rationale behind that is that some models are so big when initially loaded that they will disappear from view if we move them up to align with the build plate, causing the user to think that loading somehow failed. Big models are partially outside of the build plate, so that was seen as the threshold for it. Perhaps a better trigger could be conceived.